wiedehopf/readsb #1: selecting rtlsdr by serial number

MrBeBo999, 2021-04-19:
Hi, 

I tried to let readsb select the dongle by serial number by replacing the original string by the string found by dump1090. But this did not work. I tried some variations in syntax, but no dongle found. The original string will find the rtlsdr, but not by serial number.
Do you know the right syntax?

RECEIVER_OPTIONS="--device 0 --device-type rtlsdr --gain -10 --ppm 0"                       <-- original
RECEIVER_OPTIONS="--sdr driver=rtlsdr,serial=00001090 --gain -10 --ppm 0"               <-- from dump1090

wiedehopf, 2021-04-19:
You're using syntax used in dump978  ... it was never used in dump1090

`--device 00001090` should work.

What is the output of
```
rtl_test
```

closed 2021-04-19 by wiedehopf

MrBeBo999, 2021-04-19:
Yes, it works. I have planes in tar1090, thanks! 
Now I have to do the same trick with the other dongle and rtl_tcp ;)