wiedehopf/readsb #2: sdr device support

messschuh, 2021-06-04:
it seems to me rtl-sdr devices are supported at moment only , but is saw there are some files for other other devices like pluto-sdr...
so which devices work with readsb at moment and which are planed? i would like to get pluto or sdrplay run.  would be happy for some informations.

thx

wiedehopf, 2021-06-05:
There are no files for sdrplay ...
Not sure about the state of PLUTOSDR ... try complinging with the option ....

sdrplay and plutosdr as far as i know are more expensive than rtl-sdr models and they don't perform any better if not worse in the case of sdrplay for decoding 1090 MHz.
As such i don't have plans of supporting these two.

messschuh, 2021-06-06:
thx for reply. 

at first i used ur install script and thought it will compile for all (supportet) devices, so i tried to compiled with plutosdr as device option and it works. 
Still the /etc/default/readsb config file not works out of the box, i needed to change the device parameter for the plutosdr.

closed 2021-06-06 by messschuh

gvanem, 2022-08-31:
> As such i don't have plans of supporting these two.

Late comment, I know. But there is a SoapySDR version here https://github.com/flightaware/dump1090/tree/soapysdr
that could do it.