wiedehopf/readsb #5: Debian build fails due missing dh-systemd package on Bullseye

clyso, 2021-08-13:
As per [https://bugs.debian.org/822670](https://bugs.debian.org/822670), the dh-systemd package has been merged with the main debhelper package since version 9.20160709

For some time, a transitional dh-systemd package existed which simply depended on debhelper. However, this was removed from bullseye.

The fix should be as simple as depending on debhelper (>= 9.20160709), which available on all supported Debian releases and all supported Ubuntu releases except for xenial.

wiedehopf, 2021-08-13:
That should be better.
I don't think i need pkg-config either for the build .... removed it.
At least i could remove it from my system and still build.

Let me know if there are any issues.

wiedehopf, 2021-08-13:
Nvm ... re-adding pkg-config ...

clyso, 2021-08-13:
not able to test atm - sorry

wiedehopf, 2021-08-13:
assuming this is fixed, closing.

if you still have issues just let me know.

closed 2021-08-13 by wiedehopf