wiedehopf/readsb #9: After starting of readsb the dir /run/readsb disappear

jerzyguzenda, 2022-02-27:
Hi i just installed the readsb from the automatically script the /run/readsb directory disappear and the Tar1090 dont see server.
"Problem fetching data from the server:
Seems the decoder / receiver / backend isn't working correctly!"

When i shutdown the readsb then the /run/readsb is there and the Tor1090 show the map witch my location.
Linux Mint 20.03

readsb --device-type rtlsdr --net
readsb version: wiedehopf git: 
rtlsdr: using device #0: Terratec Cinergy T Stick RC (Rev.3) (Realtek, RTL2838UHIDIR, SN 00000001)
Found Elonics E4000 tuner
rtlsdr: enabling tuner AGC
Allocating 16 zero-copy buffers
Thx for help

wiedehopf, 2022-02-27:
doesn't linux mint have systemd?

Use the systemd service .... check the log as described in the readme for the automatic install script

jerzyguzenda, 2022-02-27:
> doesn't linux mint have systemd?
> 
> Use the systemd service .... check the log as described in the readme for the automatic install script

I'm sorry, I'm new to Linux and ADS-B.
When I start the reasdsb like this: readsb --device-type rtlsdr --net
The /run/readsb directory is gone and the tar1090 can't get data.
When I close the readsb then the readsb comes back to /run/ and the tar1090 display my loc on the map.
I installed manually and with script.

wiedehopf, 2022-02-27:
Don't start it manually then ..... i don't see the issue.

It's meant to automatically run.
You can check the log as i pointed you to: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb#issues--check-the-log

jerzyguzenda, 2022-02-27:
> Don't start it manually then ..... i don't see the issue.
> 
> It's not meant to be run manually. It starts by itself. You can check the log as i pointed you to: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb#issues--check-the-log

Thank You.
I didn't know that "It starts by itself" now i only open the tar1090 in the web browser and it works.
This is a problem because i us also CubicSDR, rtl_433 ... and now i get error because the device is used all the time.
Can i disable autostart readsb at system start and then when i need star it?

wiedehopf, 2022-02-28:
It's a systemd service called readsb.
You can find out how to stop it on this thing called internet searching for systemd services.

closed 2022-02-28 by wiedehopf