wiedehopf/readsb #12: Statistics inconsistency

chris4git, 2022-05-09:
Hi
I run "readsb --device-type rtlsdr --stats-every=1 --quiet" 
to follow the number of valid frames received per seconds but, instead of a value expected to be around 40/s, the counter rise at each step then seems to reset before rising again (see below).
I use version wiedehopf git: a30c9c3 (committed: Wed Apr 20 13:11:02 2022 0200)
Bug or misunderstanding ?

`

218 accepted with correct CRC
 398 accepted with correct CRC
    588 accepted with correct CRC
    785 accepted with correct CRC
    979 accepted with correct CRC
    1162 accepted with correct CRC
    1315 accepted with correct CRC
    1501 accepted with correct CRC
    394 accepted with correct CRC
    537 accepted with correct CRC
    720 accepted with correct CRC
    868 accepted with correct CRC
    1025 accepted with correct CRC
    313 accepted with correct CRC
    514 accepted with correct CRC
    670 accepted with correct CRC
    827 accepted with correct CRC
    987 accepted with correct CRC
    328 accepted with correct CRC
    473 accepted with correct CRC
    689 accepted with correct CRC
    782 accepted with correct CRC
    946 accepted with correct CRC
`
Chris

wiedehopf, 2022-05-09:
Call it a bug if you want.

The stats-every isn't meant for intervals shorter than couple of minutes probably.

Changing the code to support this would be annoying and i just don't need that functionality.

wiedehopf, 2022-05-09:
It seems to correctly print the timespan: 
Statistics: Mon May  9 21:32:55 2022 CEST - Mon May  9 21:33:00 2022 CEST

Yeah the stats stuff is only done in 10 second intervals.

Maybe i'll take a look if there are easy changes.

chris4git, 2022-05-09:
> It seems to correctly print the timespan: Statistics: Mon May 9 21:32:55 2022 CEST - Mon May 9 21:33:00 2022 CEST
> 
> Yeah the stats stuff is only done in 10 second intervals.
> 
> Maybe i'll take a look if there are easy changes.

thanks for your fast answer.
It would be useful to find the best place while moving the antenna in real time...

wiedehopf, 2022-05-09:
Just look at the message rate in the tar1090 webinterface for that?
Should be good enough.

chris4git, 2022-05-09:
It's small and needs a browser but you're right, I'll use it.
Thanks
Chris

closed 2022-05-09 by chris4git