wiedehopf/readsb #17: Can't compile on Alpine Linux: missing rawmemchr

Dennis14e, 2022-06-20:
When trying to compile readsb on Alpine Linux, I get the following error message:
```
#10 3.318 cc -DMODES_READSB_VERSION=\""wiedehopf git: b08247d (committed: Tue Jun 14 19:07:04 2022 0200)"\" -D_GNU_SOURCE -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -DENABLE_RTLSDR -std=c11 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2   -Wno-format-truncation -I/usr/include/ -I/usr/include/libusb-1.0  -c net_io.c -o net_io.o     
#10 3.379 net_io.c: In function 'decodeUatMessage':
#10 3.379 net_io.c:2776:17: error: implicit declaration of function 'rawmemchr'; did you mean 'memchr'? [-Werror=implicit-function-declaration]
#10 3.379  2776 |     char *eod = rawmemchr(som, '\0');
#10 3.379       |                 ^~~~~~~~~
#10 3.379       |                 memchr
#10 3.379 net_io.c:2776:17: error: initialization of 'char *' from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
#10 4.328 cc1: all warnings being treated as errors
#10 4.331 make: *** [Makefile:122: net_io.o] Error 1
```
It looks like rawmemchr is not available under musl libc.
Is it possible to use another function than rawmemchr?

wiedehopf, 2022-06-20:
fixed by cd1c9d21ddd0566b6e0d8888676ea6b23091c75c

wiedehopf, 2022-06-20:
there might be other issues with musl ....
let me know.

Dennis14e, 2022-06-21:
Seems like everything is working now, thanks!

closed 2022-06-21 by Dennis14e