wiedehopf/readsb #18: Any way to optimize the "readsb" performance on Raspberyy Pi B+

cmbahadir, 2022-07-01:
I am running readsb with fr24feeder on Raspberry Pi B+ having below Specs:

```
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 000e
Serial          : 0000000041f0e806
Model           : Raspberry Pi Model B Rev 2
```

Kernel version: `Linux raspberrypi 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux`

readsb tag/commit :  `5d238a06402a20f4ddc661e7344623ea6475a3a6`

readsb options : 

```
RECEIVER_OPTIONS="--device 0 --device-type rtlsdr --gain 30"
DECODER_OPTIONS="--preamble-threshold 0 --lat xxxxxx --lon xxxxxx --max-range 100"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005"
```

I was able to see decoded messages on --net and --interactive modes before. But after upgrading the readsb and raspberry pi kernel I haven't observed any decoded DF17 messages. I assume it is because of the performance.

Is there any way to optimize the readsb to work on Raspberry Pi B+?

wiedehopf, 2022-07-01:
You were running gain 30 before?

If you're running it as a service, you need to use viewadsb to look at the data, not run a 2nd instance.

readsb won't work super great on a raspberry pi b+ but it should be ok.
the only issue will be MLAT.

you really need to check the journalctl log as described in the readme ....

Not seeing messages with lat lon might be related to having entered the wrong lat lon?
You're pretty restrictive with the 100 nmi max range.

wiedehopf, 2022-07-01:
Check this for DF17 messages.
```
viewadsb --no-interactive --filter-DF=17
```

And this for the log ... it's in the readme and the first thing to check.
```
sudo journalctl -u readsb -e -n 30
```

cmbahadir, 2022-07-01:
Thanks for the great work you are doing and your quick response, well, I know the Raspberry Pi B+ is not the best choice, but unfortunately it is really hard to buy an Raspberry Pi these days, assuming due to the chip shortage or inflation etc. And B+ is the only available hardware I have.

But any way, here are the journalctl logs;

```
Tem 01 00:25:01 raspberrypi systemd[1]: Started readsb ADS-B receiver.
Tem 01 00:25:02 raspberrypi readsb[531]: Fri Jul  1 00:25:02 2022 +03  readsb starting up.
Tem 01 00:25:02 raspberrypi readsb[531]: readsb version: wiedehopf git: 5d238a0-dirty (committed: Wed Jun 29 22:17:05 2022 0200)
Tem 01 00:25:02 raspberrypi readsb[531]: Using lat:   41.0376, lon:   29.1175
Tem 01 00:25:02 raspberrypi readsb[531]: 30002: Raw TCP output port            30005: Beast TCP output port
Tem 01 00:25:02 raspberrypi readsb[531]: 30003: SBS TCP output ALL port        30001: Raw TCP input port
Tem 01 00:25:02 raspberrypi readsb[531]: 30004: Beast TCP input port           30104: Beast TCP input port
Tem 01 00:25:04 raspberrypi readsb[531]: rtlsdr: using device #0: Generic RTL2832U OEM (Realtek, RTL2838UHIDIR, SN )
Tem 01 00:25:06 raspberrypi readsb[531]: Found Fitipower FC0013 tuner
Tem 01 00:25:06 raspberrypi readsb[531]: rtlsdr: tuner gain set to 19.7 dB
Tem 01 00:25:06 raspberrypi readsb[531]: Allocating 16 zero-copy buffers
Tem 01 00:25:10 raspberrypi readsb[531]: FIFO dropped, suppressing this message for 30 seconds.
Tem 01 00:26:07 raspberrypi readsb[531]: Lost 139 packets (7620521.0 us) on USB, MLAT could be UNSTABLE, check sync! (ppm: -253455)(or the system clock jumped for some reason)
Tem 01 00:26:37 raspberrypi readsb[531]: Lost 126 packets (6901581.0 us) on USB, MLAT could be UNSTABLE, check sync! (ppm: -230030)(or the system clock jumped for some reason)
Tem 01 00:27:07 raspberrypi readsb[531]: Lost 80 packets (4346354.3 us) on USB, MLAT could be UNSTABLE, check sync! (ppm: -144808)(or the system clock jumped for some reason)
Tem 01 00:40:08 raspberrypi readsb[531]: SDR ppm out of specification (could cause MLAT issues) or local clock jumped / not syncing with ntp or chrony! ppm: -607
Tem 01 07:49:14 raspberrypi readsb[531]: FIFO dropped, suppressing this message for 30 seconds.
```

The gain is set to 19.4dB max.

I do not want to share the lat and lon of my location but, I entered them correctly and I can't even track the plane flying by my place. 

BTW, if I run it on my laptop with same options, it tracks pretty good with max 29nmi.

wiedehopf, 2022-07-01:
> Found Fitipower FC0013 tuner

This tuner doesn't work for 1090 MHz barely at all.
So the SDR you're using is just unsuited.

Also the CPU is just insufficient.
What else is wrong, i can't tell you.
Certainly you can't run graphs1090 on that pi, really you can't run anything else probably.

I suppose you might be better off with the super old dump1090 that use 2 MHz sampling rate instead of 2.4 MHz.

Anyhow did you have success with this exact SDR before?
If so ... just use your old configuration i can't help you.

wiedehopf, 2022-07-01:
Well it could also be that the power supply is insufficient and the SDR is not getting the proper 5 V it needs.

In any case, cool the pi if you can.
I believe there might even be a setting to allow more current to the USB port for those old RPis.

wiedehopf, 2022-07-01:
> Since when the FC0013 tuner is not supported by readsb?

It's generally incapable of proper reception on 1090 MHz. Not sure how it worked at all before, you must have a lucky sample or something.

The support is not the issue, i can't tell you why the reception is that poor.
It might just be CPU issues, can't tell you really.

You can add this, an excerpt from --help:
```
  --preamble-threshold=<40-400>                                  lower threshold --> more CPU usage (default: 58, pi zero / pi 1: 75, hot CPU 42)
```
So if you crank up the threshold, CPU usage should reduce.
Try 80, 90, 100 ... probably until you stop seeing lost stuff on the journal.

How did you compile readsb, you run the script or do it yourself?

wiedehopf, 2022-07-01:
Still running Raspbian i presume?

wiedehopf, 2022-07-01:
Oh yes and remove anything else using CPU on the RPi......

Again you might be better off just running some old dump1090 that uses 2 MHz sampling ... even dump1090-mutability you can install in raspbian does 2.4 MHz ... but maybe it has an option to do 2 MHz.
Really don't know.

cmbahadir, 2022-07-01:
> viewadsb --no-interactive --filter-DF=17

Yes.

```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

> How did you compile readsb, you run the script or do it yourself?

Built by myself.

> Again you might be better off just running some old dump1090 that uses 2 MHz sampling ... even dump1090-mutability you can install in raspbian does 2.4 MHz ... but maybe it has an option to do 2 MHz. Really don't know.

Yes, I will switch back to an old commit on dump1090.

wiedehopf, 2022-07-01:
you can try this:
```
make AIRCRAFT_HASH_BITS=10 RTLSDR=yes OPTIMIZE="-Ofast -march=native"
```

or if -Ofast causes some sort of issue (i have never tried)
```
make AIRCRAFT_HASH_BITS=10 RTLSDR=yes OPTIMIZE="-O3 -march=native"
```

wiedehopf, 2022-07-01:
Actually i don't know if march=native works on RPi 32bit ... it might not, give this a try:
```
make clean
make AIRCRAFT_HASH_BITS=10 RTLSDR=yes OPTIMIZE="-Ofast -mcpu=arm1176jzf-s -mfpu=vfp"
```

dearkey, 2023-08-03:
I have a B+ feeding FR24, FA and (adsbexchange with MLAT) for testing. The only Problem is jq in tar1090 and https://adsbexchange.com/stats.sh using 25% of the CPU every some seconds. If jq is removed from tar1090.sh and stats uninstalled everything is ok. 
[image]

wiedehopf, 2023-08-03:
Don't use adsbx stats then.
Don't use graphs.
disable the tar1090 service (interface will still work)

closed 2023-08-03 by wiedehopf

RundesBalli, 2023-08-03:
@dearkey try to stop feeding to fr24, the fr24 feederclient is predestined to consume a lot of CPU power and is often the most cpu consuming software on the whole pi. Especially on a RPI series 1. :joy:
Check with `htop` :-)

dearkey, 2023-08-03:
@wiedehopf thanks! i didn't know that the service dont need to be run @RundesBalli  fr24 use max 5% of the CPU. the worst one was glidernet in my test that need a better rpi. ;( I want to track helicopters with MLAT in my area. But i think i dont have enough data and knowlege for that now.