wiedehopf/readsb #21: readsb with mode-s beast problem

vatprcyq, 2022-08-21:
I have a modesbeast board but have problem with readsb
It  seems readsb can't set baudrate to 1000000? Test with microcom and same baudrate is normal output beast message.
But if not add --beast-baudrate=1000000 readsb will abnormal exit
Test:
 sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB1 --beast-baudrate=1000000
Mon Aug 22 02:41:40 2022 CST  readsb starting up.
readsb version: wiedehopf git: 365daf3 (committed: Fri Aug 12 02:36:00 2022 0200)
Beast cfsetispeed(/dev/ttyUSB1, 1000000): Invalid argument
Mon Aug 22 02:41:40 2022 CST  sdrOpen() failed, exiting!
Mon Aug 22 02:41:40 2022 CST  Abnormal exit.
tianyi@BPI-M2:~ $ sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000
Mon Aug 22 02:41:52 2022 CST  readsb starting up.
readsb version: wiedehopf git: 365daf3 (committed: Fri Aug 12 02:36:00 2022 0200)
Beast cfsetispeed(/dev/ttyUSB0, 1000000): Invalid argument
Mon Aug 22 02:41:52 2022 CST  sdrOpen() failed, exiting!
Mon Aug 22 02:41:52 2022 CST  Abnormal exit.
tianyi@BPI-M2:~ $
tianyi@BPI-M2:~ $ microcom -p /dev/ttyUSB0 -s 1000000
connected to /dev/ttyUSB0
Escape character: Ctrl-\
Type the escape character to get to the prompt.
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+00FFB00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+00FFB00000000000000000000000;
+00FFB00000000000000000000000;
+0001B00000000000000000000000;
+00FFB00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0001B00000000000000000000000;
+0000B00000000000000000000000;
+0001B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+00FFB00000000000000000000000;
+00FFB00000000000000000000000;
+0000B00000000000000000000000;
+0001B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0001B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+0000B00000000000000000000000;
+00FFB00000000000000000000000;

Enter command. Try 'help' for a list of builtin commands
exiting
tianyi@BPI-M2:~ $ sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0
Mon Aug 22 02:43:54 2022 CST  readsb starting up.
readsb version: wiedehopf git: 365daf3 (committed: Fri Aug 12 02:36:00 2022 0200)
Running Mode-S Beast via USB.
<3>FATAL: removeStale() interval 60.1 seconds! Trying for an orderly shutdown as well as possible!
<3> priorityTasksRun didn't run for 60.1 seconds!
<3> removeStale didn't run for 59.1 seconds!
Mon Aug 22 02:44:54 2022 CST  Abnormal exit.
tianyi@BPI-M2:~ $
 Try  stty -F /dev/ttyUSB0 ispeed 1000000 ospeed 1000000 
but not work for me.
tianyi@BPI-M2:~ $ lsusb
Bus 001 Device 005: ID 10c4:ea70 Silicon Labs CP2105 Dual UART Bridge
Bus 001 Device 004: ID 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
readsb with RTLSDR is working.

wiedehopf, 2022-08-21:
I don't have a modes beast to debug.

I'd recommend running https://github.com/flightaware/beast-splitter and connect to that using readsb.

Otherwise ... it doesn't seem you set the baudrate using the command line when using readsb ... check the help for that.

vatprcyq, 2022-08-21:
> I don't have a modes beast to debug.
> 
> I'd recommend running https://github.com/flightaware/beast-splitter and connect to that using readsb.
> 
> Otherwise ... it doesn't seem you set the baudrate using the command line when using readsb ... check the help for that.

Not working 
sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000
Mon Aug 22 03:27:07 2022 CST  readsb starting up.
readsb version: wiedehopf git: 365daf3 (committed: Fri Aug 12 02:36:00 2022 0200)
Beast cfsetispeed(/dev/ttyUSB0, 1000000): Invalid argument
Mon Aug 22 03:27:07 2022 CST  sdrOpen() failed, exiting!
Mon Aug 22 03:27:07 2022 CST  Abnormal exit.

wiedehopf, 2022-08-21:
Try the most recent commit please.

vatprcyq, 2022-08-21:
> Try the most recent commit please.

Use readsb-install.sh can't use new ver,showing ver is Aug 19 
HEAD 现在位于 37865ea better spread full trace writes after startup
rm -f *.o compat/clock_gettime/*.o compat/clock_nanosleep/*.o readsb viewadsb cprtests crctests convert_benchmark
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: 37865ea (committed: Fri Aug 19 19:18:45 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DAIRCRAFT_HASH_BITS=16 -DENABLE_RTLSDR  -g -O3 -march=native -c readsb.c -o readsb.o

Unable make manually:
make -j3 RTLSDR=yes
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: d345be4 (committed: Sun Aug 21 22:10:23 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DENABLE_RTLSDR  -g  -c readsb.c -o readsb.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: d345be4 (committed: Sun Aug 21 22:10:23 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DENABLE_RTLSDR  -g  -c globe_index.c -o globe_index.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: d345be4 (committed: Sun Aug 21 22:10:23 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DENABLE_RTLSDR  -g  -c api.c -o api.o
globe_index.c: In function ‘load_blob’:
globe_index.c:2772:25: error: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Werror=sign-compare]
 2772 |             if (end - p < compressed_len) {
      |                         ^
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: d345be4 (committed: Sun Aug 21 22:10:23 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DENABLE_RTLSDR  -g  -c minilzo/minilzo.c -o minilzo.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: d345be4 (committed: Sun Aug 21 22:10:23 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DENABLE_RTLSDR  -g  -c threadpool.c -o threadpool.o
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""wiedehopf git: d345be4 (committed: Sun Aug 21 22:10:23 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DSC16Q11_TABLE_BITS=8 -DENABLE_RTLSDR  -g  -c sdr_rtlsdr.c -o sdr_rtlsdr.o
cc1: all warnings being treated as errors
make: *** [Makefile:120：globe_index.o] 错误 1

wiedehopf, 2022-08-21:
try once more ....

You can change the automatic script to use a different branch, it's not complicated.

vatprcyq, 2022-08-22:
readsb now running with  no error but no plane showing
use beast-splitter is OK
and beast-splitter show this is Radarcape-style ?
sudo beast-splitter --serial /dev/beast --listen 37193:B
Listening on 0.0.0.0:37193
Listening on [::]:37193
serial(/dev/beast): opening port at 3000000bps
serial(/dev/beast): configured with settings: BCDfgHIjk
serial(/dev/beast): configured with settings: RCDfgHIjk
serial(/dev/beast): opening port at 1000000bps
serial(/dev/beast): configured with settings: BCDfgHIjk
serial(/dev/beast): autobaud failed, trying again (consider specifying --fixed-baud)
serial(/dev/beast): opening port at 3000000bps
serial(/dev/beast): configured with settings: BCDfgHIjk
0.0.0.0:37193: accepted a connection from 127.0.0.1:58704 with settings B
serial(/dev/beast): configured with settings: BCdfgHijk
serial(/dev/beast): opening port at 1000000bps
serial(/dev/beast): configured with settings: BCdfgHijk
serial(/dev/beast): autobaud selected 1000000 bps
serial(/dev/beast): configured with settings: RCdfgHijk
serial(/dev/beast): connected to a Radarcape-style receiver
127.0.0.1:58704: connection closed
serial(/dev/beast): configured with settings: RCDfgHIjk
0.0.0.0:37193: accepted a connection from 127.0.0.1:56562 with settings B
serial(/dev/beast): configured with settings: RCdfgHijk

vatprcyq, 2022-08-22:
sudo beast-splitter --serial /dev/beast --listen 37193:B
./readsb   --net-connector=127.0.0.1,37193,beast_in  --net-only --interactive
is OK

wiedehopf, 2022-08-22:
I don't know what hardware you have.

Honestly i'm not sure i have interest in further debugging this ... without the hardware it's kinda complicated.
Don't know what it needs beyond selecting 1000000 baud.

wiedehopf, 2022-08-22:
Ok ... shot in the dark, try the most recent version of readsb with the baudrate specified.

wiedehopf, 2024-09-15:
Install the sid branch like this:
 ```
 wget -O readsb-install.sh https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh
 bash readsb-install.sh sid
 ```
 
 Add this option, it might produce a lot of output:
 ```
--devel=debugSerial
```

The issue is probably is that it's reading from the device and not getting anything (and blocking on the read).
I set it non blocking with that debug option but i'm not sure that's a good idea really.
So ... with non blocking it might not crash but i don't think it'll work either.

dchaplinsky, 2024-09-16:
Hi @wiedehopf, thanks for prompt reply.

no real change here:
```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
[2024-09-16 09:06:42.095 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: e6c2edb (committed: Sun Sep 15 22:10:44 2024 0200)
Running Mode-S Beast via USB.
06:06:42.111 event count 0 wait_ms 20
<3>FATAL: removeStale() interval 60.1 seconds! Trying for an orderly shutdown as well as possible!
misc thread: threadSignalJoin timed out after 30.0 seconds, undefined behaviour may result!
upkeep thread: threadSignalJoin timed out after 15.0 seconds, undefined behaviour may result!
decode thread: threadSignalJoin timed out after 7.5 seconds, undefined behaviour may result!
<3>FATAL: thread upkeep could not be joined, calling abort()!
Aborted
```

dchaplinsky, 2024-09-16:
I can see some binary data with microcom tho:
[image]

wiedehopf, 2024-09-16:
This does tell us something though.

Try again please, sid branch has been changed.
Hopefully using the correct setting now to at least have a timeout for the reads from serial.

If you just want to get going i'd recommend this software which is better tested due to the amount of users and persons maintaining it having the hardware in question: <https://github.com/flightaware/beast-splitter>
To avoid having to compile it, you can install it from the flightaware repository like this:
```
wget -O /tmp/piaware-repo.deb https://flightaware.com/adsb/piaware/files/packages/pool/piaware/f/flightaware-apt-repository/flightaware-apt-repository_1.2_all.deb
sudo apt purge -y piaware-repository &>/dev/null
sudo rm -f /etc/apt/sources.list.d/piaware-*.list
sudo dpkg -i /tmp/piaware-repo.deb
sudo apt update
sudo apt install -y beast-splitter
```

You can set it to listen on some port and have readsb connect via TCP to get the data from it.

```
readsb --net --net-connector 127.0.0.1,37232,beast_in`
beast-splitter --serial /dev/beast --listen 37232:R
```

Possibly with this timeout i just put into the code readsb will work normally when possibly before it wasn't working properly when there was nothing being received.
beast-splitter should allow you at least to confirm if it's working.

dchaplinsky, 2024-09-16:
```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
[2024-09-16 13:05:11.028 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 17fac8d (committed: Mon Sep 16 09:49:29 2024 0200)
Running Mode-S Beast via USB.
10:05:11.043 event count 0 wait_ms 20
10:05:11.063 serial read ... fd: 0 maxbytes: 262140
<3>FATAL: removeStale() interval 60.1 seconds! Trying for an orderly shutdown as well as possible!
```

I've tried beast splitter yesterday, but I'm not sure if it works properly. Not so many planes around me (I'm from Ukraine). I had the hope that there might be some over south of Belarus, but apparently those is no-fly zone:

```
# sudo beast-splitter --serial /dev/ttyUSB0 --listen 37232:R --fixed-baud 1000000
Listening on 0.0.0.0:37232
Listening on [::]:37232
serial(/dev/ttyUSB0): opening port at 1000000bps
serial(/dev/ttyUSB0): configured with settings: BCDfgHIjk
0.0.0.0:37232: accepted a connection from 127.0.0.1:52404 with settings R
serial(/dev/ttyUSB0): configured with settings: BCdfGHijk
```

```
Beast TCP input: Connection established: 127.0.0.1 port 37232
Beast TCP input: No data or heartbeat received for 120 seconds, reconnecting: 127.0.0.1 port 37232
Beast TCP input: Connection established: 127.0.0.1 port 37232
Beast TCP input: No data or heartbeat received for 120 seconds, reconnecting: 127.0.0.1 port 37232
Beast TCP input: Connection established: 127.0.0.1 port 37232
Beast TCP input: No data or heartbeat received for 120 seconds, reconnecting: 127.0.0.1 port 37232
Beast TCP input: Connection established: 127.0.0.1 port 37232
Beast TCP input: No data or heartbeat received for 120 seconds, reconnecting: 127.0.0.1 port 37232
Beast TCP input: Connection established: 127.0.0.1 port 37232
Beast TCP input: No data or heartbeat received for 120 seconds, reconnecting: 127.0.0.1 port 37232
Beast TCP input: Connection established: 127.0.0.1 port 37232
```

wiedehopf, 2024-09-16:
Might just be that the receiver isn't working.
fr24 somewhat has the best coverage.
Your reception range will depend heavyily on geography and where the antenna is.
This should give you some hints: <https://github.com/wiedehopf/tar1090?tab=readme-ov-file#heywhatsthatcom-range-outline>

Anyhow this gives me some idea possibly:
```
10:05:11.063 serial read ... fd: 0 maxbytes: 262140
```
I don't think fd == 0 can be correct ... i suspect it's a race with the logic initializing the SDR.
Could you run readsb-install.sh sid once more and try again, ty.

wiedehopf, 2024-09-16:
Oh actually i think that's still not correct ... let me change the code once more.

Code changed on sid branch.

dchaplinsky, 2024-09-16:
Looks the same to my eye:

```
pi@adsb:~ $ sudo lsof | grep ttyUSB
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
readsb    4152                            root    5u      CHR      188,0      0t0        273 /dev/ttyUSB0
readsb    4152 4153 readsb                root    5u      CHR      188,0      0t0        273 /dev/ttyUSB0
readsb    4152 4154 readsb                root    5u      CHR      188,0      0t0        273 /dev/ttyUSB0
readsb    4152 4155 readsb                root    5u      CHR      188,0      0t0        273 /dev/ttyUSB0
readsb    4152 4156 readsb                root    5u      CHR      188,0      0t0        273 /dev/ttyUSB0
readsb    4152 4157 readsb                root    5u      CHR      188,0      0t0        273 /dev/ttyUSB0
```

```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
[2024-09-16 13:53:53.487 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: d0b3135 (committed: Mon Sep 16 12:29:22 2024 0200)
Running Mode-S Beast via USB.
10:53:53.504 event count 0 wait_ms 20
10:53:53.524 serial read ... fd: 0 maxbytes: 262140
<3>FATAL: removeStale() interval 60.1 seconds! Trying for an orderly shutdown as well as possible!
misc thread: threadSignalJoin timed out after 30.0 seconds, undefined behaviour may result!
upkeep thread: threadSignalJoin timed out after 15.0 seconds, undefined behaviour may result!
decode thread: threadSignalJoin timed out after 7.5 seconds, undefined behaviour may result!
<3>FATAL: thread upkeep could not be joined, calling abort()!
Aborted
```

Let me try newest commit

dchaplinsky, 2024-09-16:
Newer build:

```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --beast-baudrate=1000000 --devel=debugSerial
[2024-09-16 13:58:40.931 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 2a5e8b9 (committed: Mon Sep 16 12:56:14 2024 0200)
10:58:40.947 event count 0 wait_ms 0
Running Mode-S Beast via USB.
10:58:40.947 event count 0 wait_ms 0
10:58:40.947 serial: creating socket client ...
10:58:40.947 serial: creating socket client ... done
10:58:40.947 serial read ... fd: 5 maxbytes: 262140
10:58:40.947 serial read return value: 64
10:58:40.962 event count 0 wait_ms 20
10:58:40.982 serial read ... fd: 5 maxbytes: 262076
<3>FATAL: removeStale() interval 60.1 seconds! Trying for an orderly shutdown as well as possible!
misc thread: threadSignalJoin timed out after 30.0 seconds, undefined behaviour may result!
upkeep thread: threadSignalJoin timed out after 15.0 seconds, undefined behaviour may result!
decode thread: threadSignalJoin timed out after 7.5 seconds, undefined behaviour may result!
<3>FATAL: thread upkeep could not be joined, calling abort()!
Aborted
```

dchaplinsky, 2024-09-16:
Wait! Different (default) baudrate gave me this:
```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial
[2024-09-16 14:03:11.717 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 2a5e8b9 (committed: Mon Sep 16 12:56:14 2024 0200)
11:03:11.729 event count 0 wait_ms 0
11:03:11.729 event count 0 wait_ms 0
11:03:11.729 event count 0 wait_ms 0
11:03:11.729 event count 0 wait_ms 0
11:03:11.730 event count 0 wait_ms 0
11:03:11.731 event count 0 wait_ms 0
11:03:11.731 event count 0 wait_ms 0
11:03:11.731 event count 0 wait_ms 0
11:03:11.732 event count 0 wait_ms 0
11:03:11.732 event count 0 wait_ms 0
Running Mode-S Beast via USB.
11:03:11.732 event count 0 wait_ms 0
11:03:11.733 serial: creating socket client ...
11:03:11.733 serial: creating socket client ... done
11:03:11.733 serial read ... fd: 5 maxbytes: 262140
11:03:11.733 serial read return value: 16
<3>High load: modesNetPeriodicWork() elapsed1/2/3/interval 13328/0/0/1 ms, suppressing for 5 seconds!
<3> priorityTasksRun didn't run for 13.3 seconds!
<3> removeStale didn't run for 12.3 seconds!
11:03:25.062 event count 0 wait_ms 20
*00e89072dfea05;
hex:  8fbeec   CRC: 8fbeec fixed bits: 0 decode: reject unknown hex
RSSI:    -18.6 dBFS   reduce_forward: 0
receiverId: e003af09-e8dd-77da
receiverTime:               1447605841.67us
utcTime: 11:03:11.733 epoch: 1726484591.733
DF:  0 addr: 8fbeec   VS:0 CC:0 SL:7 RI:1 AC:4210
 Short Air-Air Surveillance
  ICAO Address:  8FBEEC (mode_s)
  Air/Ground:    airborne?

11:03:25.083 serial read ... fd: 5 maxbytes: 262140
```

dchaplinsky, 2024-09-16:
And then:

```
# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial
[2024-09-16 14:03:11.717 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 2a5e8b9 (committed: Mon Sep 16 12:56:14 2024 0200)
11:03:11.729 event count 0 wait_ms 0
11:03:11.729 event count 0 wait_ms 0
11:03:11.729 event count 0 wait_ms 0
11:03:11.729 event count 0 wait_ms 0
11:03:11.730 event count 0 wait_ms 0
11:03:11.731 event count 0 wait_ms 0
11:03:11.731 event count 0 wait_ms 0
11:03:11.731 event count 0 wait_ms 0
11:03:11.732 event count 0 wait_ms 0
11:03:11.732 event count 0 wait_ms 0
Running Mode-S Beast via USB.
11:03:11.732 event count 0 wait_ms 0
11:03:11.733 serial: creating socket client ...
11:03:11.733 serial: creating socket client ... done
11:03:11.733 serial read ... fd: 5 maxbytes: 262140
11:03:11.733 serial read return value: 16
<3>High load: modesNetPeriodicWork() elapsed1/2/3/interval 13328/0/0/1 ms, suppressing for 5 seconds!
<3> priorityTasksRun didn't run for 13.3 seconds!
<3> removeStale didn't run for 12.3 seconds!
11:03:25.062 event count 0 wait_ms 20
*00e89072dfea05;
hex:  8fbeec   CRC: 8fbeec fixed bits: 0 decode: reject unknown hex
RSSI:    -18.6 dBFS   reduce_forward: 0
receiverId: e003af09-e8dd-77da
receiverTime:               1447605841.67us
utcTime: 11:03:11.733 epoch: 1726484591.733
DF:  0 addr: 8fbeec   VS:0 CC:0 SL:7 RI:1 AC:4210
 Short Air-Air Surveillance
  ICAO Address:  8FBEEC (mode_s)
  Air/Ground:    airborne?

11:03:25.083 serial read ... fd: 5 maxbytes: 262140
<3>FATAL: priorityTasksRun() interval 60.7 seconds! Trying for an orderly shutdown as well as possible!
<3>lockThreads() probably hung on decode
<3>FATAL: removeStale() interval 61.7 seconds! Trying for an orderly shutdown as well as possible!
11:03:25.083 serial read return value: 16
<3>High load: modesNetPeriodicWork() elapsed1/2/3/interval 81279/0/0/21 ms, suppressing for 5 seconds!
<3> priorityTasksRun didn't run for 81.3 seconds!
<3> removeStale didn't run for 80.3 seconds!
[2024-09-16 14:04:46.367 EEST] Abnormal exit. uptime:  0d  0h  1m 34.651s
```

wiedehopf, 2024-09-16:
Well that's much better.

New version that hopefully doesn't block.

dchaplinsky, 2024-09-16:
Ping me when you want me to do another test run.

wiedehopf, 2024-09-16:
let's go ahead and do another test.

If this works, it will print a lot of info now, you can just stop it after 10 or 15 seconds and try again without --devel=debugSerial

dchaplinsky, 2024-09-16:
That is with debugSerial:
```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial
[2024-09-16 15:06:18.347 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 50cc9de (committed: Mon Sep 16 13:30:59 2024 0200)
Running Mode-S Beast via USB.
12:06:18.381 event count 0 wait_ms 20
12:06:18.381 serial: creating socket client ...
12:06:18.382 serial: creating socket client ... done
12:06:18.382 serial read return value: 1148
12:06:18.402 event count 0 wait_ms 20

 Survelliance, Identity Reply
  ICAO Address:  9345D4 (mode_s)
  Air/Ground:    airborne?
  Squawk:        4330

*83046cae03b7f7ea6c36966841db;
hex:  01fc4f   CRC: 01fc4f fixed bits: 0 decode: reject unknown hex
RSSI:    -20.5 dBFS   reduce_forward: 0
receiverId: 2ad03ec3-bb62-4396
receiverTime:               4020866435.58us
utcTime: 12:06:18.382 epoch: 1726488378.382
DF: 16 addr: 01fc4f   VS:0 SL:0 RI:8 AC:3246 MV:03B7F7EA6C3696
 Long Air-Air ACAS
  ICAO Address:  01FC4F (mode_s)
  Air/Ground:    airborne?
  Baro altitude:             100000 ft

*23c1eb7d3841e5;
hex:  8db015   CRC: 8db015 fixed bits: 0 decode: reject unknown hex
RSSI:    -20.2 dBFS   reduce_forward: 0
receiverId: 2ad03ec3-bb62-4396
receiverTime:               4044409329.42us
utcTime: 12:06:18.382 epoch: 1726488378.382
DF:  4 addr: 8db015   FS:3 DR:24 UM:15 AC:2941
 Survelliance, Altitude Reply
  ICAO Address:  8DB015 (mode_s)
  Air/Ground:    ground

*83107b8fb80ea144164487d6db5c;
hex:  ea10a1   CRC: ea10a1 fixed bits: 0 decode: reject unknown hex
RSSI:    -19.8 dBFS   reduce_forward: 0
receiverId: 2ad03ec3-bb62-4396
receiverTime:               4237237568.08us
utcTime: 12:06:18.382 epoch: 1726488378.382
DF: 16 addr: ea10a1   VS:0 SL:0 RI:0 AC:7055 MV:B80EA144164487
 Long Air-Air ACAS
  ICAO Address:  EA10A1 (mode_s)
  Air/Ground:    airborne?

*2160c59e610322;
hex:  b10dba   CRC: b10dba fixed bits: 0 decode: reject unknown hex
RSSI:    -19.8 dBFS   reduce_forward: 0
receiverId: 2ad03ec3-bb62-4396
receiverTime:               4272597138.08us

utcTime: 12:06:18.382 epoch: 1726488378.382
DF:  0 addr: 72ebc1   VS:0 CC:1 SL:7 RI:1 AC:2220
 Short Air-Air Surveillance
  ICAO Address:  72EBC1 (mode_s)
  Air/Ground:    airborne?

*2fdc80a713dd3c;
hex:  041bd8   CRC: 041bd8 fixed bits: 0 decode: reject unknown hex
RSSI:    -18.6 dBFS   reduce_forward: 0
receiverId: 2ad03ec3-bb62-4396
receiverTime:               4467545726.75us
utcTime: 12:06:18.382 epoch: 1726488378.382
[... 272 lines cut ...]
  Air/Ground:    airborne?
  Baro altitude:              95100 ft

12:06:18.421 serial read return value: 0
Get client socket error failed.
Shutdown client socket failed.
Serial client closed unexpectedly, exiting!
12:06:18.421 event count 1 wait_ms 20
Serial client closed unexpectedly, exiting!
[2024-09-16 15:06:18.428 EEST] Abnormal exit. uptime:  0d  0h  0m 0.081s
```
Prints a lot indeed and then dies almost immediatelly.

With no flag it looks like this:
```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0
[2024-09-16 15:07:31.037 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 50cc9de (committed: Mon Sep 16 13:30:59 2024 0200)
Running Mode-S Beast via USB.
Get client socket error failed.
Shutdown client socket failed.
Serial client closed unexpectedly, exiting!
[2024-09-16 15:07:31.080 EEST] Abnormal exit. uptime:  0d  0h  0m 0.044s
```

dchaplinsky, 2024-09-16:
Loud and clear! Let me try interactive mode!

[image]

Yeah, it is probably noise, as antenna is not even connected.

dchaplinsky, 2024-09-16:
When in --interactive mode, does it record caught vehicles somewhere?

wiedehopf, 2024-09-16:
No.
It'll display them for 60 seconds.

Really you want to modify /etc/default/readsb and make sure the systemd service is working.
Then use the tar1090 webinterface.

Before you do that, can you run update to sid branch once more and run this:
```
sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --quiet --stats --devel=debugSerial
```
Maybe 10 or 20 seconds should be enough.
Just to give me some info if epoll_wait works.

dchaplinsky, 2024-09-16:
Here you go:

```
root@adsb:~# sudo readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --quiet --stats --devel=debugSerial
invoked by: readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --quiet --stats --devel=debugSerial
[2024-09-16 17:15:12.444 EEST] readsb starting up.
readsb version: 3.14.1639 wiedehopf git: 66857d3 (committed: Mon Sep 16 15:02:37 2024 0200)
Running Mode-S Beast via USB.
14:15:12.557 serial: creating socket client ...
14:15:12.559 serial: creating socket client ... done
14:15:12.560 serial read return value: 32
14:15:12.662 serial read return value: 0
14:15:12.764 serial read return value: 0
14:15:12.866 serial read return value: 0
14:15:12.968 serial read return value: 0
14:15:13.073 serial read return value: 0
14:15:13.173 serial read return value: 0
14:15:13.274 serial read return value: 0
14:15:13.374 serial read return value: 0
14:15:13.474 serial read return value: 0
14:15:13.575 serial read return value: 0
14:15:13.675 serial read return value: 0
14:15:13.775 serial read return value: 0
14:15:13.875 serial read return value: 0
14:15:13.976 serial read return value: 0
14:15:14.076 serial read return value: 0
14:15:14.176 serial read return value: 0
14:15:14.277 serial read return value: 0
14:15:14.377 serial read return value: 0
14:15:14.477 serial read return value: 0
14:15:14.578 serial read return value: 0
14:15:14.678 serial read return value: 0
14:15:14.778 serial read return value: 0
14:15:14.879 serial read return value: 0
14:15:14.979 serial read return value: 0
14:15:15.079 serial read return value: 0
14:15:15.180 serial read return value: 0
14:15:15.280 serial read return value: 0
14:15:15.380 serial read return value: 0
14:15:15.481 serial read return value: 0
14:15:15.581 serial read return value: 0
14:15:15.681 serial read return value: 0
14:15:15.782 serial read return value: 0
14:15:15.882 serial read return value: 0
14:15:15.982 serial read return value: 0
14:15:16.083 serial read return value: 0
14:15:16.183 serial read return value: 0
14:15:16.283 serial read return value: 0
14:15:16.384 serial read return value: 0
14:15:16.484 serial read return value: 0
14:15:16.584 serial read return value: 0
14:15:16.685 serial read return value: 0
14:15:16.785 serial read return value: 0
14:15:16.885 serial read return value: 0
14:15:16.986 serial read return value: 0
14:15:17.086 serial read return value: 0
14:15:17.186 serial read return value: 0
14:15:17.287 serial read return value: 0
14:15:17.387 serial read return value: 0
14:15:17.487 serial read return value: 0
14:15:17.588 serial read return value: 0
14:15:17.688 serial read return value: 0
14:15:17.788 serial read return value: 0
14:15:17.889 serial read return value: 0
14:15:17.989 serial read return value: 0
14:15:18.089 serial read return value: 0
14:15:18.190 serial read return value: 0
14:15:18.290 serial read return value: 0
14:15:18.390 serial read return value: 0
14:15:18.491 serial read return value: 0
[... 301 lines cut ...]
    0 accepted with 1-bit error repaired
2 total usable messages
0 surface position messages received
0 airborne position messages received
0 global CPR attempts with valid positions
0 global CPR attempts with bad data
  0 global CPR attempts that failed the range check
  0 global CPR attempts that failed the speed check
0 global CPR attempts with insufficient data
0 local CPR attempts with valid positions
  0 aircraft-relative positions
  0 receiver-relative positions
0 local CPR attempts that did not produce useful positions
  0 local CPR attempts that failed the range check
  0 local CPR attempts that failed the speed check
0 CPR messages that look like transponder failures filtered
0 non-ES altitude messages from ES-equipped aircraft ignored
0 unique aircraft tracks
0 aircraft tracks where only one message was seen
CPU load: 0.2%
  0 ms for demodulation
  0 ms for reading from USB
  59 ms for network input and background tasks
[2024-09-16 17:15:45.037 EEST] Normal exit. uptime:  0d  0h  0m 32.594s
```

wiedehopf, 2024-09-16:
Yeah that's not quite conclusive as there were no messages coming in i suppose.

Can you recompile one last time and run it for a couple of minutes, it should be less spammy now :)
Not sure what you changed when it was emitting all those messages, you got an antenna connected now?

dchaplinsky, 2024-09-16:
Nope, not yet. Will try it tonight, quite busy with my work now and antenna mounting require some manual intervention.

closed 2024-09-16 by wiedehopf

dchaplinsky, 2024-09-16:
Connected the antenna and changed the service settings like this:

```
RECEIVER_OPTIONS="--device-type modesbeast --beast-serial=/dev/ttyUSB0"
DECODER_OPTIONS="--lat 50.3679557298612 --lon 30.610656947644898 --max-range 450 --write-json-every 1"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005"
JSON_OPTIONS="--json-location-accuracy 2 --range-outline-hours 24"
```

I dropped also the ppm and gain parameters which was there. ppm I think is rtlsdr only, but not sure about gain.

wiedehopf, 2024-09-16:
Yeah that's good.
```
journalctl -u readsb -e
```
That looking reasonable?

 /tar1090 working?
 
 Also note you might want to remove the location just because. But it doesn't really matter i suppose.
 
 There is also GPS spoofing / GPS jamming going on you can check
 https://map.adsbexchange.com/mlat-map/
 https://globe.adsbexchange.com/
 Now it's not known which range these receivers have, but yeah i wouldn't expect a lot of signals.
 Flight paths in Belarus are probably 200 nmi away from Ukraine, at least what's visible on most trackers.

dchaplinsky, 2024-09-16:
I wasn't using the correct coordinates just in case, but yeah, good catch, didn't mean to post even those.

Previously I've been able to track flights in Belarus, but it was rather high floor, now I just stick antenna out of my window.

/tar1090 works good, but nothing is recorded.

dchaplinsky, 2024-09-17:
24 hours so far, I think it is working, but catching nothing, which is understandable

[image]

wiedehopf, 2024-09-17:
When you have time can you maybe re-run the readsb-install.sh sid and add to /etc/default/readsb with the other options:
```
--devel=debugSerial
```

And restart the service.
It shouldn't spam the log too much.
And the log should give me confirmation that the code is correct.

dchaplinsky, 2024-09-17:
Did that, will report if anything fancy caught

dchaplinsky, 2024-09-18:
I can see a lot of records like this:
```
Sep 18 18:42:56 adsb readsb[25507]: 15:42:56.919 event count 1 wait_ms 1000
Sep 18 18:42:56 adsb readsb[25507]: 15:42:56.919 serial read return value: 16
Sep 18 18:43:04 adsb readsb[25507]: 15:43:04.765 event count 1 wait_ms 1000
Sep 18 18:43:04 adsb readsb[25507]: 15:43:04.766 serial read return value: 16
Sep 18 18:43:04 adsb readsb[25507]: 15:43:04.845 event count 1 wait_ms 1000
Sep 18 18:43:04 adsb readsb[25507]: 15:43:04.845 serial read return value: 23
Sep 18 18:43:10 adsb readsb[25507]: 15:43:10.949 event count 1 wait_ms 1000
Sep 18 18:43:10 adsb readsb[25507]: 15:43:10.949 serial read return value: 16
Sep 18 18:43:20 adsb readsb[25507]: 15:43:20.632 event count 1 wait_ms 1000
Sep 18 18:43:20 adsb readsb[25507]: 15:43:20.632 serial read return value: 16
Sep 18 18:43:28 adsb readsb[25507]: 15:43:28.766 event count 1 wait_ms 1000
Sep 18 18:43:28 adsb readsb[25507]: 15:43:28.766 serial read return value: 16
Sep 18 18:43:28 adsb readsb[25507]: 15:43:28.893 event count 1 wait_ms 1000
Sep 18 18:43:28 adsb readsb[25507]: 15:43:28.893 serial read return value: 16
Sep 18 18:44:27 adsb readsb[25507]: 15:44:27.488 event count 1 wait_ms 1000
Sep 18 18:44:27 adsb readsb[25507]: 15:44:27.488 serial read return value: 17
Sep 18 18:44:29 adsb readsb[25507]: 15:44:29.646 event count 1 wait_ms 1000
Sep 18 18:44:29 adsb readsb[25507]: 15:44:29.646 serial read return value: 17
Sep 18 18:44:33 adsb readsb[25507]: 15:44:33.689 event count 1 wait_ms 1000
Sep 18 18:44:33 adsb readsb[25507]: 15:44:33.689 serial read return value: 16
```

Apparent from them:
```
Sep 17 23:46:20 adsb systemd[1]: Started readsb ADS-B receiver.
Sep 17 23:46:20 adsb readsb[25507]: invoked by: /usr/bin/readsb --device-type modesbeast --beast-serial=/dev/ttyUSB0 --devel=debugSerial --lat 50.3679557298612 --lon 30.610656947644898 --max-range 450 --write-json-every 1 --net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 2 --range-outline-hours 24 --write-json /run/readsb --quiet
Sep 17 23:46:20 adsb readsb[25507]: [2024-09-17 23:46:20.659 EEST] readsb starting up.
Sep 17 23:46:20 adsb readsb[25507]: readsb version: 3.14.1640 wiedehopf git: 4b18685 (committed: Mon Sep 16 17:11:25 2024 0200)
Sep 17 23:46:20 adsb readsb[25507]: Using lat: xxxxxx, lon: xxxxxx
Sep 17 23:46:20 adsb readsb[25507]: 30002: Raw TCP output port
Sep 17 23:46:20 adsb readsb[25507]: Running Mode-S Beast via USB.
Sep 17 23:46:20 adsb readsb[25507]: 30005: Beast TCP output port
Sep 17 23:46:20 adsb readsb[25507]: 30003: SBS TCP output ALL port
Sep 17 23:46:20 adsb readsb[25507]: 30001: Raw TCP input port
Sep 17 23:46:20 adsb readsb[25507]: 30004: Beast TCP input port
Sep 17 23:46:20 adsb readsb[25507]: 30104: Beast TCP input port
Sep 17 23:46:21 adsb readsb[25507]: 20:46:21.682 serial: creating socket client ...
Sep 17 23:46:21 adsb readsb[25507]: 20:46:21.682 serial: creating socket client ... done
```

wiedehopf, 2024-09-18:
Thank you that pretty much confirms that epoll_wait is working fine, really no reason it would not i suppose.

Appreciate sticking with it taking the time to fix this. (i pretty obviously broke this at some point ... oh well)

dchaplinsky, 2024-09-18:
Thanks for your time and patience!