wiedehopf/readsb #22: Couldn't compile on Alpine Linux (musl)

Dennis14e, 2022-09-26:
Hi,

currently I get following error message compiling on Alpine Linux:

```
In file included from readsb.c:54:
readsb.h:357:18: error: missing binary operator before token "("
  357 | #if __GNUC_PREREQ(8,0)
     |                  ^
make: *** [Makefile:117: readsb.o] Error 1
```

Also see [here](https://github.com/flighttrackr/docker-readsb/actions/runs/3125859763/jobs/5070695621#step:7:373).

wiedehopf, 2022-09-26:
Which gcc is that?

wiedehopf, 2022-09-26:
Check the possible fix please.

Dennis14e, 2022-09-27:
gcc version is 11.2.1, error still exists.

```
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""3.14.1593 wiedehopf git: 30f19f6 (committed: Mon Sep 26 23:53:18 2022 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DENABLE_RTLSDR -I/usr/include/ -I/usr/include/libusb-1.0 -g -c readsb.c -o readsb.o 
In file included from readsb.c:54: 
readsb.h:357:43: error: missing binary operator before token "(" 
| #if defined __GNUC_PREREQ && __GNUC_PREREQ(8,0) 
      |                                           ^
make: *** [Makefile:117: readsb.o] Error 1
```

wiedehopf, 2022-09-27:
And now?

closed 2022-09-27 by wiedehopf

Dennis14e, 2022-09-27:
Works again now, thanks!