wiedehopf/readsb #23: trace.json and history_x.json not being created

adam-ahern, 2022-12-04:
### Hardware Info
- Raspberry Pi 3B+ 1GB
- 0bda:2838 Realtek Semiconductor Corp. RTL2838 DVB-T

### Software Info
readsb version
```
readsb version: 3.14.1595 wiedehopf git: a25f8ea (committed: Fri Nov 11 18:19:32 2022 0100)
```

Program options
```
cat /etc/default/readsb
# readsb configuration
# This is sourced by /etc/systemd/system/default.target.wants/readsb.service as
# daemon startup configuration.

RECEIVER_OPTIONS="--device 0 --device-type rtlsdr --gain -10 --ppm 0"
DECODER_OPTIONS="--lat xxx --lon yyy --max-range 450 --write-json-every 1"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005"
JSON_OPTIONS="--json-location-accuracy 2 --range-outline-hours 24 --json-trace-interval 1"
OTHER_OPTIONS="--heatmap-dir /run/readsb --heatmap 5 --write-state /run/readsb --write-globe-history /run/readsb"
```

I installed readsb using the scripted method outlined in the README. fr24feed is reading from the beast output. 

### Problem
trace.json and history.json are not being written to the `/run/readsb` directory. History files are being written to `/run/tar1090/`. 

```
pi@raspberrypi:~ $ ls -l /run/readsb/
total 40
drwxr-xr-x 3 readsb nogroup   60 Dec  4 18:01 2022
-rw-r--r-- 1 readsb nogroup  276 Dec  4 18:44 aircraft.binCraft.zst
-rw-r--r-- 1 readsb nogroup 1611 Dec  4 18:44 aircraft.json
drwxr-xr-x 2 readsb nogroup 3720 Dec  4 18:44 internal_state
-rw-r--r-- 1 readsb nogroup 5803 Dec  4 18:44 outline.json
-rw-r--r-- 1 readsb nogroup  251 Dec  4 18:01 receiver.json
-rw-r--r-- 1 readsb nogroup 5877 Dec  4 18:44 stats.json
-rw-r--r-- 1 readsb nogroup 3128 Dec  4 18:44 stats.prom
-rw-r--r-- 1 readsb nogroup  336 Dec  4 18:44 status.json
-rw-r--r-- 1 readsb nogroup  496 Dec  4 18:44 status.prom

pi@raspberrypi:~ $ ls -l /run/tar1090/
...
-rw-r--r-- 1 tar1090 nogroup  298 Dec  4 18:43 history_1670179396928.json
-rw-r--r-- 1 tar1090 nogroup  298 Dec  4 18:43 history_1670179404939.json
...
```

Am I missing a configuration option or doing something otherwise incorrect to not be getting these files?

wiedehopf, 2023-02-27:
why are you expecting the same things in /run/tar1090 and /run/readsb ?

For traces see: https://github.com/wiedehopf/tar1090#0800-destroy-sd-card

gvanem, 2023-05-29:
Can I butt in here (instead of opening another issue). I'm trying to understand how this `/?pTacks/` URL gets handled by the
readsb program since I want to handle such a request myself. 

AFAICS, this tracks-feature depends on a lot of:
 * `traces/XX/trace_recent_YYYYYY.json` 
 * `traces/XX/trace_full_YYYYYY.json` 

files written by `generateTraceJson()`. Possibly gzip/zstd compressed, right?
But what's the relation and use of a `trace_recent_YYYYYY.json` and `trace_full_YYYYYY.json`?
The `html/script.js` is hard to follow since I do not know much JavaScript.

wiedehopf, 2023-05-29:
No it uses the data produced by tar1090.sh in that repo.
It basically builds its own history.
This is the origin of my tar1090 webinterface.

At some point i'd like to move that generation of data to readsb but it's not been a priority as it works perfectly fine.

gvanem, 2023-05-30:
> No, pTracks (and the 45 min history on non globe-index installs) uses the data produced by tar1090.sh in that repo.

OK, thanks.

closed 2025-08-11 by wiedehopf