wiedehopf/readsb #29: Uncompiled code

palhaland, 2023-03-26:
There are a couple of ifs that always will evaluate to 0, should these be removed, replaced with a macro, or removed the `0 &&` so that they are compiled in?

A lot of them are debug (fprintf).

```
globe_index.c:776:    if (0 && a->addr == TRACE_FOCUS)
globe_index.c-777-        fprintf(stderr, "mw: %.0f, perm: %.0f, count: %d\n",
globe_index.c-778-                ((int64_t) a->trace_next_mw - (int64_t) now) / 1000.0,
--
globe_index.c:843:        if (0 && oldSize != newSize) {
globe_index.c-844-            fprintf(stderr, "%06x size mismatch when replacing aircraft data, aborting!\n", source->addr);
globe_index.c-845-            return -1;
--
globe_index.c:1218:            if (0 && focus) {
globe_index.c-1219-                time_t nowish = state->timestamp/1000;
globe_index.c-1220-                struct tm utc;
--
globe_index.c:1250:            if (0 && focus) {
globe_index.c-1251-                time_t nowish = state->timestamp/1000;
globe_index.c-1252-                struct tm utc;
--
globe_index.c:1286:                    if (0 && focus) {
globe_index.c-1287-                        fprintf(stderr, "k: %d %d %d %d\n", k, (int) (st->baro_alt / _alt_factor), st->baro_alt_valid, st->on_ground);
globe_index.c-1288-                    }
--
globe_index.c:1327:        if (0 && elapsed > 30 * 60 * 1000 && (state->on_ground || !state->baro_alt_valid || (state->baro_alt_valid && state->baro_alt / _alt_factor < max_leg_alt))) {
globe_index.c-1328-            double distance = greatcircle(
globe_index.c-1329-                    (double) state->lat * 1e-6,
--
globe_index.c:1485:    if (0 && ca->reader_count > 1) {
globe_index.c-1486-        fprintf(stderr, "ca->reader_count %d\n", ca->reader_count);
globe_index.c-1487-    }
--
globe_index.c:1713:        if (0 && Modes.verbose) {
globe_index.c-1714-            fprintf(stderr, "%06x deleting %d chunks\n", a->addr, deletedChunks);
globe_index.c-1715-        }
--
globe_index.c:1729:        if (0 && Modes.verbose) {
globe_index.c-1730-            fprintf(stderr, "%s%06x tracePrune a->trace_current: %d\n",
globe_index.c-1731-                    ((a->addr & MODES_NON_ICAO_ADDRESS) ? "." : ". "), a->addr, SFOUR * deleteFs);
--
globe_index.c:2014:        if (0 && lastChunk && memcmp(zstd_magic, lastChunk->compressed, sizeof(zstd_magic)) == 0) {
globe_index.c-2015-            // recompress finished buffer
globe_index.c-2016-            recompressStateChunk(lastChunk, passbuffer);
--
globe_index.c:2043:        if (0 && Modes.json_dir) {
globe_index.c-2044-            char path[1024];
globe_index.c-2045-            snprintf(path, 1024, "%s/tracechunk_samples/%06x", Modes.json_dir, a->addr);
--
globe_index.c:2174:        if (0 && Modes.verbose) {
globe_index.c-2175-            fprintf(stderr, "len %d max %d\n", a->trace_current_len, a->trace_current_max);
globe_index.c-2176-        }
--
json_out.c:1264:        if (0 && (p + 2000) >= end) {
json_out.c-1265-            int used = p - buf;
json_out.c-1266-            alloc *= 2;
--
json_out.c:1609:    if (0 && a->addr == TRACE_FOCUS) {
json_out.c-1610-        fprintf(stderr, "%06x sprintCache: %d points firstRecent starting %d (firstRecentCache starting %d, max %d)\n", a->addr, tb.len - firstRecent, firstRecent, firstRecentCache, Modes.traceCachePoints);
json_out.c-1611-    }
--
json_out.c:1687:        if (0 && state->timestamp < lastTs) {
json_out.c-1688-            fprintf(stderr, "%06x trace timestamps wrong order: %.3f %.3f i: %d k: %d %s\n", a->addr, lastTs / 1000.0, state->timestamp / 1000.0, i, k, stringStart);
json_out.c-1689-        }
--
json_out.c:1786:            if (0 && a->addr == TRACE_FOCUS) {
json_out.c-1787-                fprintf(stderr, "%06x using tCache starting with tCache->firstRecentCache %d stateIndex %d\n", a->addr, tCache->firstRecentCache, start);
json_out.c-1788-            }
--
mode_s.c:1807:    if (0 && mm->cpr_valid && mm->cpr_decoded) {
[... 82 lines cut ...]
--
track.c:2421:                if (0 && greatcircle(a->mlat_lat, a->mlat_lon, mm->decoded_lat, mm->decoded_lon, 1) > 5000) {
track.c-2422-                    a->mlat_pos_valid.source = SOURCE_INVALID;
track.c-2423-                }
--
track.c:2429:                if (0 && a->pos_reliable_valid.source > SOURCE_MLAT) {
track.c-2430-                    fprintf(stderr, "%06x: %d\n", a->addr, mm->reduce_forward);
track.c-2431-                }
--
track.c:2480:                        if (0 && a->position_valid.last_source > SOURCE_INDIRECT && a->position_valid.source == SOURCE_INVALID) {
track.c-2481-                            mm->decoded_lat = a->lat;
track.c-2482-                            mm->decoded_lon = a->lon;
--
track.c:3612:    if (0 && a->addr == Modes.cpr_focus)
track.c-3613-        fprintf(stderr, "%06x: position_bad %.1f %.1f %u %u\n", a->addr, a->pos_reliable_odd, a->pos_reliable_even, mm->cpr_lat, mm->cpr_lon);
track.c-3614-}
--
track.h:640:    if (0 && Modes.position_persistence > reliable && reliable > 1 && (a->addr & MODES_NON_ICAO_ADDRESS || a->addrtype == ADDR_TISB_ICAO || a->addrtype == ADDR_ADSR_ICAO)) {
track.h-641-        reliable += 1; // require additional reliability for non-icao hex addresses
track.h-642-    }
--
net_io.c:1310:    if (Modes.debug_ping && 0) {
net_io.c-1311-        char uuid[64]; // needs 36 chars and null byte
net_io.c-1312-        sprint_uuid(c->receiverId, c->receiverId2, uuid);
```

wiedehopf, 2023-03-26:
I don't see an issue here.

palhaland, 2023-03-27:
No issues, I just wanted to contribute and needed a decision on what action to do. If there are no issues here, then I assume the associated code can be removed? If you don't mind I'll make a pull request on it.

Alternatively everything that is related to debug can use a compile-flag if it is going to be included.

(Started running clang --analyze on the code)

wiedehopf, 2023-03-27:
what I mean is the code will stay. As it is.

wiedehopf, 2023-03-27:
Please don't get me wrong if there is a bug or you want to add some feature, i'm open to pull requests.

I'd even consider other (manually found) cleanups or adding comments where i consider it a positive change.

But stuff static code analyzers are just not something i'm interested in.
I mean if the static analyzer finds a bug or some memory unsafe thing i'll consider it.
But this change wouldn't change the compiled code.
I left this stuff in for future work or debugging of the respective functions.
I'd then remove && 0 to reactivate an old behaviour or turn on the debug output.
Having it all behind a general debug flag would make the log unreadable pretty much.
I suppose one could add sensible debug flags for several logic parts, but i'm not sure it's worth it.

And in regards to code readability: Anyone serious about working on the code will see this stuff as deactivated.
Doing a block comment or ifdef might be cleaner but i like this style as it's super quick for me to do.

The reality is, since i forked readsb i've done 99.9% of code changes.
So mostly i'll accept performance improvements / bug fixes / features added.
Features need to be opt in via command line if they use significant resources.