wiedehopf/readsb #33: Option '--throttle'

gvanem, 2023-04-19:
I've ported *readsb* to Windows and now trying to use it with the `--ifile <IQ-file>` and `--throttle` options is rather confusing.

Apart from `epoll()` and `O_BINARY` issue with binary data, it was quite easy to port to Windows.
First I generated IQ-data using `rtl_sdr -f 1090M -s 2.4M -n 1E8 samples-bin.iq`,
and running `readsb.exe --device-type ifile --ifile samples-bin.iq` shows:
```
*8d4785c1225c91b81a0820cd3afd;
hex:  4785c1   CRC: 000000 fixed bits: 0 decode: ok
RSSI:     -7.5 dBFS   reduce_forward: 1
Score: 1400
receiverTime:                    52492.08us
utcTime: 05:45:04.359 epoch: 1681883104.359
DF: 17 AA:4785C1 CA:5 ME:225C91B81A0820
 Extended Squitter Aircraft identification and category (4)
  ICAO Address:  4785C1 (adsb_icao)
  Air/Ground:    airborne
  Ident:         WIF8F
  Category:      A2

*a000061e8db00000000000a91648;
hex:  4785c1   CRC: 4785c1 fixed bits: 0 decode: ok
RSSI:     -7.4 dBFS   reduce_forward: 1
Score: 1000
receiverTime:                    57065.08us
utcTime: 05:45:04.363 epoch: 1681883104.363
DF: 20 addr: 4785c1   FS:0 DR:0 UM:0 AC:1566 MB:8DB00000000000
 Comm-B, Altitude Reply
  Comm-B format: BDS4,0 Selected vertical intention
  ICAO Address:  4785C1 (mode_s)
  Air/Ground:    airborne?
  Baro altitude:               8950 ft
  MCP selected altitude:       7008 ft
...
```

But adding the `--throttle` as in `readsb.exe --throttle --device-type ifile --ifile samples-bin.iq`, gives:
```
SDR type '(null)' not recognized; supported SDR types are:
  rtlsdr
  modesbeast
  gnshulc
  ifile
  none
invoked by: readsb.exe --throttle --device-type ifile --ifile samples-bin.iq
```
But it runs fine without throttling.

Experimenting with adding it last `readsb.exe --device-type ifile --ifile samples-bin.iq --throttle`,
takes 45 sec to complete. Makes sense (since a `readsb.exe --device-type ifile --ifile samples-bin.iq`, takes 10 sec to complete).

Why can't I have the `--throttle` option first?

BTW. using `--interactive --throttle ...`, `readsb.exe` crashes inside `interactiveCleanup()` since my PD-curses was not started.

wiedehopf, 2023-04-19:
I don't think it's an option for ifile .... not sure.

wiedehopf, 2023-04-19:
In general ... can you test the exact same command on linux?
Then i know if there is an actual issue in readsb and you know if it's due to the port.

gvanem, 2023-04-19:
> I don't think it's an option for ifile .... not sure.

The `readsdb --help` page says it is:
```
ifile-specific options, use with --ifile:
  --ifile=<path>                                                 Read samples from given file ('-' for stdin)
  --iformat=<type>                                               Set sample format (UC8, SC16, SC16Q11)
  --throttle                                                     Process samples at the original capture speed
```

> In general ... can you test the exact same command on linux?

I have only *Windows Subsystem for Linux* which is incapable of  building `readsb`. Even a `sudo apt install zlib1g` fails?!

wiedehopf, 2023-04-19:
Run a VM.

gvanem, 2023-04-19:
I don't have one. Besides I'm not friend with Linux and my port did nothing to effect the cmdline parsing.
IMHO there are far too many options.

gvanem, 2023-04-19:
OK. I managed to build and run in WSL. 
* This throttles:  `./readsb --device-type ifile --ifile samples-bin.iq --throttle`
* This does not:  `./readsb --throttle --device-type ifile --ifile samples-bin.iq`

I.e. It shows the same in the 1st case:
```
<3> priorityTasksRun didn't run for 1681887115.5 seconds!
<3> removeStale didn't run for 1681887114.5 seconds!
<3> priorityTasksRun didn't run for 5.0 seconds!
....
```
And this `SDR type '(null)' not recognized:` in the 2nd case. Not a porting issue AFAICS.

closed 2023-04-19 by wiedehopf

gvanem, 2023-04-19:
> The webinterface looks something like this: https://globe.adsb.lol/

This is awesome. But no planes over my town Bergen (Norway).

>  you'd have to put that into a fork on your github and maintain it yourself.

That's what I'm doing; readsb works extremely good with the 2.4MS/s decoder. I'm hoping to port that over to my own dump1090 port. But hope to avoid all the thread-usage in readsb.

wiedehopf, 2023-04-19:
readsb is meant to do a lot of stuff.
Thread usage isn't a bad thing, isn't there a pthread implementation that works on windows?

I've fixed the errors and hiccups with ifile ... at least most of it.
It still pauses when it's done before it exits but that's an issue for later.

If you've already ported readsb ... not sure why you'd use dump1090 to be honest.
readsb is just so much better ....

You can feed the aggrgators to add data.
Possibly check the sdr-enthusiasts discord, it's a nice place: https://discord.gg/sTf9uYF

https://globe.adsb.lol/
https://globe.adsb.fi/
https://globe.adsbexchange.com/ (i worked on this until the owner sold it and now it still has good data but is no better than fr24 / fa .... might even start blocking soon no one knows)

gvanem, 2023-04-19:
I think I'll stick to https://planefinder.net/. I only wish they'd support distance-rings.