wiedehopf/readsb #37: Send GPS Signal to reads

madjuju, 2023-05-01:
How can l send my location to reads ? L what to run it while l move

RundesBalli, 2023-05-01:
https://gpsd.gitlab.io/gpsd/

From `readsb --help`:
```
--net-connector=<ip,port,protocol>
Establish connection, (...) Protocols: (...) gpsd_in, (...)
```

So you have to set up gpsd, check which port gpsd provides your position from your gps antenna, and then put into the readsb config: `sudo nano /etc/default/readsb`
```
--net-connector 127.0.0.1,<gpsd-port>,gpsd_in
```

madjuju, 2023-05-01:
Yes it's OK 
So can l send my location to tar1090 ?

RundesBalli, 2023-05-01:
If I remember correctly, tar1090 loads your position with a full-reload of the site.

wiedehopf, 2023-05-01:
Will autoupdate while on the page.
There is a setting in the webpage menu to update gps location or sth ...

RundesBalli, 2023-05-01:
[image]

madjuju, 2023-05-01:
l have found "update location" cleared my cache but nothing appear

madjuju, 2023-05-01:
it's OK gaps was not fully configured. 

Thank you very much

kiwibenis, 2024-09-30:
> Does gpsd integration in readsb only move the marker on the map or does it also update the station location that is sent to adsbexchange for MLAT?

it only updates your Map-Location

kiwibenis, 2024-09-30:
> [image]

be aware:
https://github.com/wiedehopf/readsb/issues/40#issuecomment-2382722138