wiedehopf/readsb #41: Compile error in function get_seed

Dennis14e, 2023-06-24:
Using Alpine 3.18.2

```
cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""3.14.1604 wiedehopf git: 0059b62 (committed: Sun Jun 18 07:12:59 2023 0200)"\" -Wdate-time -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DENABLE_RTLSDR -I/usr/include/ -I/usr/include/libusb-1.0 -g  -c util.c -o util.o
util.c: In function 'get_seed':
util.c:215:110: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
  215 |     unsigned int seed = (uint64_t) time.tv_sec ^ (uint64_t) time.tv_nsec ^ (((uint64_t) getpid()) << 16) ^ (((uint64_t) pthread_self()) << 10);
      |                                                                                                              ^
cc1: all warnings being treated as errors
make: *** [Makefile:125: util.o] Error 1
```

wiedehopf, 2023-06-24:
```c
unsigned int seed = (uint64_t) time.tv_sec ^ (uint64_t) time.tv_nsec ^ (((uint64_t) getpid()) << 16) ^ (((uint64_t) (uintptr_t) pthread_self()) << 10);
```
Can you try this line pls.

Dennis14e, 2023-06-24:
Yes, this fixes this.

closed 2023-06-25 by wiedehopf