wiedehopf/readsb #46: Will it work on RISC-V?

streamofstars, 2023-10-15:
Has someone already tried that or perhaps the project creator knows that from the top of their head: **will this compile and work on RISC-V CPUs?**

My use case is basically building the smallest and the cheapest receiver. I can live with stats and local website radar view disabled (it will just feed to another instance of readsb on a server elsewhere). Some cheap ARM SBCs like old Raspberry Pi Zero come to mind but nowadays a lot of smaller and cheaper RISC-V devices pop up on the market, perhaps those can be used? In that case, do you have any general advice, like e.g. must have more then x MB of RAM or at least x cores, or specific chip manufacturer, etc.?

wiedehopf, 2023-10-15:
It's just C..... 
If you can run a linux distro on it it'll run readsb pretty much ...

wiedehopf, 2023-10-15:
No i don't have any further advice that's up to you to test whatever usecase you're after.

closed 2024-05-11 by wiedehopf