wiedehopf/readsb #49: alternating gain

jorisrobijn, 2023-11-15:
At a local airport we have an ADS-B receiver, but it is swamped by strong signals. Of course we could reduce the (optimized) gain, but then planes further away will not be received.
I don't know how invasive the gain switching is, but I was wondering if the following would work. When a signal has been received that's too strong (or perhaps simply constantly), the gain is reduced for 1.5 seconds, then switches back to normal gain for 7 seconds, after which it repeats. Maybe these times should be somewhat randomized. 
How far to reduce during this 1.5 seconds low-gain window? Well maybe "a lot" (30 dB less? to minimum?) might just work, or perhaps the user could set it. 
I was thinking of some gain regulator, and while that works for approaching aircraft that get louder and louder, that doesn't work well for airports where aircraft can startup nearby unexpectedly. So I think a fixed reduced gain would be best.
I chose the times based on this https://aviation.stackexchange.com/questions/78952/what-is-the-broadcast-rate-of-an-ads-b-message .
What do you think about this?

RundesBalli, 2023-11-15:
Use this: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-gain-optimization-for-readsb-and-dump1090-fa

jorisrobijn, 2023-11-17:
I don't think you read what I wrote.

RundesBalli, 2023-11-17:
How often do you want to restart readsb? :-D

Just leave it with autogain and the results will be the best on average for the day.

wiedehopf, 2023-11-17:
He is asking for changing gain during operation.

I have no plans on working on such a feature, but i'll leave the issue open in case anyone wants to work on coding something like that.
I'm not 100% sure if librtlsdr supports changing gain while the devices is streaming data.
Interrupting the data stream is not feasible due to MLAT clock.

Considering in pretty much all circumstances aggregators are gonna have far away traffic from another feeder i'm somewhat of the opinion to just focus on local aircraft and be done with it.
Or run 2 SDRs ... or upgrade to an airspy mini for example which has larger dynamic range.

wiedehopf, 2024-05-15:
For your information, i recently added something to change gain while readsb is running.

```
echo 43.9 > /run/readsb/setGain
```

closed 2025-08-11 by wiedehopf