wiedehopf/readsb #56: Flight Aware ProStick receives 0 messages

SenorFusion, 2024-02-14:
Problem: I have a Flight Aware ProStick Plus that I cannot get to work. Apologies if this is the wrong place to ask, you can point me somewhere else.

Setup:
-PI Zero 2 W
-Clean fresh Install of Dietpi
-Clean Fresh Install of all of wiedehopf install scripts (readsb, tar1090, timelapse, graphs, auto-gain)

What I have tried:
-Reboot
-Reinstall readsb script
-Wipe SD and reinstall OS from scratch and run all scripts again

What works:
-Switching to a standard RTL-SDR dongle works and planes and messages start flowing _immediately_ upon plugging it in.

What doesn't work:
-Using the exact same setup, antenna, pi, everything - just swapping SDR dongle the FA stick receives absolutely nothing.

Is my hardware bad, or is there something else required to make FA work?

Thank you!

Below is the log output of readsb with the FA stick. It seems to find it and connect correctly:

```
root@DietPi:~# sudo journalctl --no-page -u readsb
Feb 14 19:37:06 DietPi systemd[1]: Started readsb.service - readsb ADS-B receiver.
Feb 14 19:37:06 DietPi readsb[421]: invoked by: /usr/bin/readsb --device 0 --device-type rtlsdr --gain -10 --ppm 0 --lat 35.155777 --lon -85.316484 --max-range 450 --write-json-every 1 --net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 2 --range-outline-hours 24 --write-json /run/readsb --quiet
Feb 14 19:37:06 DietPi readsb[421]: [2024-02-14 19:37:06.285 GMT] readsb starting up.
Feb 14 19:37:06 DietPi readsb[421]: readsb version: 3.14.1607 wiedehopf git: d2f5725 (committed: Fri Feb 9 15:00:10 2024 0100)
Feb 14 19:37:06 DietPi readsb[421]: Using lat:   35.1558, lon:  -85.3165
Feb 14 19:37:06 DietPi readsb[421]: 30002: Raw TCP output port
Feb 14 19:37:06 DietPi readsb[421]: 30005: Beast TCP output port
Feb 14 19:37:06 DietPi readsb[421]: 30003: SBS TCP output ALL port
Feb 14 19:37:06 DietPi readsb[421]: 30001: Raw TCP input port
Feb 14 19:37:06 DietPi readsb[421]: 30004: Beast TCP input port
Feb 14 19:37:06 DietPi readsb[421]: 30104: Beast TCP input port
Feb 14 19:37:06 DietPi readsb[421]: rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001000)
Feb 14 19:37:06 DietPi readsb[421]: Detached kernel driver
Feb 14 19:37:06 DietPi readsb[421]: Found Rafael Micro R820T tuner
Feb 14 19:37:06 DietPi readsb[421]: rtlsdr: enabling tuner AGC
Feb 14 19:37:07 DietPi readsb[421]: Allocating 16 zero-copy buffers

```

wiedehopf, 2024-02-14:
FA stick probably bad.

You could try lower gain just because but i doubt it'll make a diff.

SenorFusion, 2024-02-14:
Well that is frustrating - it has never even been used.

Ok, thank you for the info!

wiedehopf, 2024-02-14:
```
systemctl stop readsb
timeout 5 readsb --no-interactive --quiet --device-type rtlsdr --stats --gain 38
```
I guess you can try this.

wiedehopf, 2024-02-14:
Just to explain maybe, the output will show some statistics on noise and stuff.

There is a very unlikely possibility that the LNA is getting overloaded completely.
But yeah it's very unlikely, a defect is the more likely scenario.

SenorFusion, 2024-02-15:
Thank you for the advice. I ran that, below is the outpout. I don't know what to make of it. 

```
root@DietPi:~# systemctl stop readsb
root@DietPi:~# timeout 5 readsb --no-interactive --quiet --device-type rtlsdr --stats --gain 38
invoked by: readsb --no-interactive --quiet --device-type rtlsdr --stats --gain 38
[2024-02-15 01:53:45.435 GMT] readsb starting up.
readsb version: 3.14.1607 wiedehopf git: d2f5725 (committed: Fri Feb 9 15:00:10 2024 0100)
rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001000)
Detached kernel driver
Found Rafael Micro R820T tuner
rtlsdr: tuner gain set to 38.6 dB
Allocating 16 zero-copy buffers
[2024-02-15 01:53:50.433 GMT] Caught SIGTERM, shutting down...
Reattached kernel driver

Statistics: Thu Feb 15 01:53:45 2024 GMT - Thu Feb 15 01:53:50 2024 GMT
Local receiver:
  9568256 samples processed
  0 samples dropped
  0 samples lost
  0 Mode A/C messages received
  166447 Mode-S message preambles received
    113539 with bad message format or invalid CRC
    52903 with unrecognized ICAO address
    5 accepted with correct CRC
    0 accepted with 1-bit error repaired
  -16.5 dBFS noise power
  -13.8 dBFS mean signal power
  -12.2 dBFS peak signal power
  0 messages with signal power above -3dBFS

 Phase stats
         3        4        5        6        7
     93386    93386    99161    99161    81372
         4        5        6        7        8
         2        0        2        0        1

5 total usable messages
0 surface position messages received
0 airborne position messages received
0 global CPR attempts with valid positions
0 global CPR attempts with bad data
  0 global CPR attempts that failed the range check
  0 global CPR attempts that failed the speed check
0 global CPR attempts with insufficient data
0 local CPR attempts with valid positions
  0 aircraft-relative positions
  0 receiver-relative positions
0 local CPR attempts that did not produce useful positions
  0 local CPR attempts that failed the range check
  0 local CPR attempts that failed the speed check
0 CPR messages that look like transponder failures filtered
0 non-ES altitude messages from ES-equipped aircraft ignored
2 unique aircraft tracks
0 aircraft tracks where only one message was seen
CPU load: 21.9%
  628 ms for demodulation
  513 ms for reading from USB
  0 ms for network input and background tasks
[2024-02-15 01:53:50.654 GMT] Normal exit.
```

SenorFusion, 2024-02-15:
I left the FA running for a few hours and got the following. You can obviously tell where I was using the RTL and switched to the FA

[image]
[image]

But After you pointed me towards too much gain maybe being the issue I found some of your other documentation and advice on the FA forum.

I ran your check and got: `Percentage of strong messages: 80.769`

I ran `sudo readsb-gain 20.7`

and the message rate on the FA has gone up to to ~5/second. So the hardware at least partly works. Still less than the RTL though.

I will continue testing lower gain settings - Thank you!

SenorFusion, 2024-02-15:
Just to update, I had your Autogain installed, but rather than wait for it over days, I ran the [2 minute version found here](https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-gain-optimization-for-readsb-and-dump1090-fa) and it worked perfectly.

FA stick is now getting 40-50 messages/second with ~3% strong messages.

```
root@DietPi:~# sudo autogain1090
Increasing gain to 36.4 (0.000% messages >-3dB)
root@DietPi:~# for i in {0..30}; do sudo autogain1090; sleep 120; done
The decoder hasn't been running long enough, wait a bit!
Increasing gain to 38.6 (0.000% messages >-3dB)
Increasing gain to 40.2 (0.000% messages >-3dB)
No gain change needed, percentage of messages >-3dB is in nominal range. (3.210%)
```

I'll continue to play with, but you can close this issue. Thank you for the support and amazingly good documentation!

wiedehopf, 2024-02-15:
The noise figures seem very high:
  -16.5 dBFS noise power
  -13.8 dBFS mean signal power
  -12.2 dBFS peak signal power

You can try placing the antenna further from other electronics.
Or there is just strong interference that might be hard to escape at your location.

SenorFusion, 2024-02-15:
That was before I figured out the game adjustment.

But it is still just setting on my desk by my computers and monitors with the stock RTL-SDR antenna right now. I have a proper ADS-B antenna and outdoor installation coming soon.

After adjustment here is how last night went. Still could probably use some tuning, but working much better.

[image]

wiedehopf, 2024-02-15:
That looks better indeed, the noise power at -16.5 was with a gain of 38 from the command line run.
That's roughly the gain you're running now.

And yeah HDMI cables can have huge noise output and even moving the antenna a bit will change the results a lot.
It is really rare to get no signals at all though ... at least i mentioned it was possible it was just noise in the first reply.

closed 2024-05-15 by wiedehopf