wiedehopf/readsb #58: Short documentation of how readsb is utilizing mlat

Exeu, 2024-03-09:
Hello,
this is more an understanding topic rather than an issue.
I am wondering how readsb is getting mlat information.
I am running readsb with tar1090.
So in tar1090 I can observe that there are several aircrafts tracked using mlat.
If I read directly from the json port of readsb I can see the mlat fields. 
Where readsb is getting this information from? For that purpose it would need to talk to different services or talk to different readsb instances in order to do triangulation? 
I can not really find this in code but maybe I am also wrong or overseeing this.

Thank you in advance!
Great work

wiedehopf, 2024-03-09:
MLAT results are being pushed into readsb by mlat-client via network most likely.

But i can't guess where you get MLAT results into your readsb that would be on you to know :)

closed 2024-03-10 by Exeu

Telecomeansa, 2026-08-18:
Good morning, put in readsb --net-sbs-in-port 30106 but in the multilateration server in sudo nano /etc/systemd/system/mlat-server.service 
--client-listen 31090 \ 
--filtered-basestation-listen 31003 \ 
--basestation-listen 31004 \ 
--filtered-basestation-connect 127.0.0.1:30107 , but I can't see anything in tar1090 in a multilateration process

wiedehopf, 2026-08-18:
Don't mulipost or I ban