wiedehopf/readsb #60: Feature Request: preserve outlines.json after restart

dxmekch, 2024-03-19:
Hello:)

I wonder if it's possible to implement an option (e.g. `--range-outline-preserve=`) to retain the outlines.json (similar to 
 `--write-state=`) after a restart of readsb or the device. This would probably make a helpful addition to the existing option 
 `--range-outline-hours=` and would prevent that the outlines are reset after a power loss or restart of the device (e.g. to apply updates).

wiedehopf, 2024-03-19:
write-state already preserves the outline.

dxmekch, 2024-03-20:
I just restarted my readsb and it appears that the traces are saved but the outlines are definitely not.

wiedehopf, 2024-03-20:
Please restart the service again and provide me with logs from startup to exit.

dxmekch, 2024-03-20:
I guess I figured it out, it seems like we can't point to the same dir as --write-json, `/run/readsb` in my case, for state storage, this dir seems not to be persistent.

`..... first start with state enabled / directory not backed by persistent storage`

I now changed the dir and it's working.

`Mar 20 17:48:33 pi3 readsb[27288]: loading state .....
Mar 20 17:48:33 pi3 readsb[27288]: actual range outline, read bytes: 368644
Mar 20 17:48:33 pi3 readsb[27288]:  .......... done, loaded 44 aircraft in 0.018 seconds!`

closed 2024-03-27 by wiedehopf