wiedehopf/readsb #62: How to run readsb without writing json at all to disk?

Exeu, 2024-03-19:
Hi,
i am running two instances of readsb on the same rpi with different rtl devices.
This works fine but apparently they both try to write json files to the same directory (/run/readsb).
So sometimes one instances seems to to sth. in this directory causing the other one to trigger an error.

Since i dont need this json stuff, is it somehow possible to deactivate this json writing at all?
I am only interested in the received data from my antenna, and readsb is just forwarding the data using beast protocol to another centralized instance.

thanks

wiedehopf, 2024-03-19:
You're running via the automatic install i assume ... the json writing is defined in the systemd service file. /lib/system/systemd/readsb.service

As far as command line options go ... just get rid of any you don't think you need.
Consult readsb --help for details.

Oh didn't read all of it ... you probably copied the systemd service without giving them separated directories which isn't too complicated if you look at the service file.

wiedehopf, 2024-03-19:
Also note you can't do MLAT on combined data.

Exeu, 2024-03-19:
Thanks for the answer.
I will have a look.

Yes for mlat I have separate mlat clients running for each readsb. Works fine

wiedehopf, 2024-03-19:
I wouldn't usually suggest this, but maybe you might like the ultrafeeder container, containers make it easy to deconflict such things: https://sdr-enthusiasts.gitbook.io/ads-b/

Exeu, 2024-03-19:
I already had look into this but it seemed to me a bit too big for my usecase. I just use a few of those containers to consolidate the data but without the ultrafeeder itself. For me feeding is a nice side-effect but my primary goal is to store my received data in a huge database where i can do heavy aggregations on :)

Can you recommend a good mlat server? 
Here in Germany currently adsbexchange seems to produce the best results followed by planespotters.
But maybe there are others out which have more peers?

wiedehopf, 2024-03-19:
Nope.
I like adsb.lol but they don't have as much MLAT coverage.

closed 2024-03-19 by wiedehopf