wiedehopf/readsb #69: Build In Gentoo:  add "-ltinfo" for ncurses to successfully link

jlpoolen, 2024-05-15:
I attempted to build readsb in Gentoo and ran into this error:

	jlpoole@ryzdesk /usr/local/src/readsb $ make
	cc -o readsb readsb.o argp.o anet.o interactive.o mode_ac.o mode_s.o comm_b.o json_out.o net_io.o crc.o demod_2400.o uat2esnt/uat2esnt.o uat2esnt/uat_decode.o stats.o cpr.o icao_filter.o track.o util.o fasthash.o convert.o sdr_ifile.o sdr_beast.o sdr.o ais_charset.o globe_index.o geomag.o receiver.o aircraft.o api.o minilzo.o threadpool.o  -pthread -lpthread -lm -lrt -lzstd -lz -lncurses
	/usr/libexec/gcc/x86_64-pc-linux-gnu/ld: interactive.o: undefined reference to symbol 'stdscr'
	/usr/libexec/gcc/x86_64-pc-linux-gnu/ld: /lib64/libtinfo.so.6: error adding symbols: DSO missing from command line
	collect2: error: ld returned 1 exit status
	make: *** [Makefile:158: readsb] Error 1
	jlpoole@ryzdesk /usr/local/src/readsb $
	
I spent several hours tracking down what may be causing the problem in Gentoo.  I then successfully built this project in Debian thereby confirming my suspicions that Gentoo's build of the ncurses library differed from Debian's.    Gentoo's ncurses library has the symbol "**stdscr**" [The Standard Screen (https://manpages.org/stdscr/3)] undefined, see the "U" below.  The fact that the symbol is undefined causes the linker to halt.

		jlpoole@ryzdesk /usr/local/src/readsb $ nm -gD /lib64/libncurses.so.6.4 |nl|grep stdscr
		311  0000000000014af0 T _nc_stdscr_of
		428                   U stdscr
		jlpoole@ryzdesk /usr/local/src/readsb $

Compare with the symbol output from Debian's library where there follows an index, i.e. "@NCURSES6_TINFO_5.0.19991023":

     jlpoole@debian1:~$ nm -gD /usr/lib/x86_64-linux-gnu/libncursesw.so.6.4 |grep stdscr
                      U stdscr@NCURSES6_TINFO_5.0.19991023
     jlpoole@debian1:~$ 

I learned that is may be possible in Gentoo Linux that the ncurses library is broken out into two library files -- a fact which proved to be true.

The symbol, **stdscr**, is defined within library **tinfo**:

		jlpoole@ryzdesk /usr/local/src/readsb $ ls -la /lib64/libncurses.so.6.4 /lib64/libtinfo.so.6.4
		-rwxr-xr-x 1 root root 171872 May  5 15:50 /lib64/libncurses.so.6.4
		-rwxr-xr-x 1 root root 262920 May  5 15:50 /lib64/libtinfo.so.6.4
		jlpoole@ryzdesk /usr/local/src/readsb $ nm -gD /lib64/libtinfo.so.6.4 |grep stdscr
		0000000000040e30 B stdscr
		jlpoole@ryzdesk /usr/local/src/readsb $
Using the command with the option "--libs", /usr/bin/ncurses6-config --libs, below, I was able to identify the missing library: **tinfo**.  

		jlpoole@ryzdesk /usr/local/src/readsb $ /usr/bin/ncurses6-config --libs
		-L/usr/lib64 -lncurses -ltinfo
		jlpoole@ryzdesk /usr/local/src/readsb

Oddly enough, though, Debian indicates that the tinfo library should be referenced, too, but as we see below, it was not necessary to reference **tinfo** to cause readsb to successfully link:

     jlpoole@debian1:~$  /usr/bin/ncurses6-config --libs
     -lncurses -ltinfo
     jlpoole@debian1:~$ 

Here's the linker command from my Debian build (with surrounding commands for context) showing that the "-ltinfo" was not included and thus not needed:

     cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""3.14.1623 wiedehopf git: 3faa6d1 (committed: Sat May 11 10:13:09 2024 0200)"\" -Wdate-time -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DAIRCRAFT_HASH_BITS=15 -DENABLE_RTLSDR -I/usr/include/  -g  -c readsb.c -o readsb.o
     cc -o readsb readsb.o argp.o anet.o interactive.o mode_ac.o mode_s.o comm_b.o json_out.o net_io.o crc.o demod_2400.o uat2esnt/uat2esnt.o uat2esnt/uat_decode.o stats.o cpr.o icao_filter.o track.o util.o fasthash.o convert.o sdr_ifile.o sdr_beast.o sdr.o ais_charset.o globe_index.o geomag.o receiver.o aircraft.o api.o minilzo.o threadpool.o sdr_rtlsdr.o -Wl,-z,relro -Wl,-z,now  -pthread -lpthread -lm -lrt -lzstd -lz -lncurses -lrtlsdr -lusb-1.0 
     rm -f viewadsb

Accordingly, I took the link command that failed (above in the Gentoo console) and appended " -ltinfo" and ran same and the linker, and thus the build with "make", succeeded.

		jlpoole@ryzdesk /usr/local/src/readsb $ /usr/bin/ncurses6-config --libs
		-L/usr/lib64 -lncurses -ltinfo
		jlpoole@ryzdesk /usr/local/src/readsb $ cc -o readsb readsb.o argp.o anet.o interactive.o mode_ac.o mode_s.o comm_b.o json_out.o net_io.o crc.o demod_2400.o uat2esnt/uat2esnt.o uat2esnt/uat_decode.o stats.o cpr.o icao_filter.o track.o util.o fasthash.o convert.o sdr_ifile.o sdr_beast.o sdr.o ais_charset.o globe_index.o geomag.o receiver.o aircraft.o api.o minilzo.o threadpool.o -pthread -lpthread -lm -lrt -lzstd -lz -lncurses -ltinfo
		jlpoole@ryzdesk /usr/local/src/readsb $ make
		rm -f viewadsb
		cp readsb viewadsb
		jlpoole@ryzdesk /usr/local/src/readsb

While my manual addition helped me successfully build the project, I leave this Issue open for @wiedehopf  to close and/or consider further.  I can see that my adding -ltinfo, there could be a risk of the linker aborting with "No such file or directory" if in that particular build of Debian the ncurses libraries are not split.  I can see in Gentoo a package, were one built for readsb, to query "usr/bin/ncurses6-config --libs"  and then assure the results are added to the linker command.

wiedehopf, 2024-05-15:
It would seem it's already linking against libtinfo ... or libncurses is linked against libtinfo.
Furthermore Debian has a libtinfo-dev package, but it's provided by libncurses-dev which needs to be installed anyhow to compile with ncurses.
Thus it should be fairly safe to add.

closed 2024-05-15 by wiedehopf

Dennis14e, 2024-05-21:
@wiedehopf Commit https://github.com/wiedehopf/readsb/commit/75eddd7382f9889cdda107731e7cb340facd4d46 breaks building on Alpine Linux (there is no libtinfo)

wiedehopf, 2024-05-21:
```pkg-config --libs ncurses```

does this return something sensible on alpine?

wiedehopf, 2024-05-21:
Or just clone and make and see if that works @Dennis14e 
on debian it doesn't hurt and it should work on gentoo i assume ....

jlpoolen, 2024-05-21:
> @wiedehopf Commit [75eddd7](https://github.com/wiedehopf/readsb/commit/75eddd7382f9889cdda107731e7cb340facd4d46) breaks building on Alpine Linux (there is no libtinfo)

@Dennis14e  can you provide the command line that breaks the build as well as the ensuing error message?

Dennis14e, 2024-05-21:
@jlpoolen 
> @Dennis14e can you provide the command line that breaks the build as well as the ensuing error message?

https://github.com/flighttrackr/docker-readsb/actions/runs/9175599780/job/25228918045?pr=198#step:7:858

Command:
```
make readsb RTLSDR=yes BLADERF=yes HACKRF=yes SOAPYSDR=yes
```

Error:
```
28.38 /usr/lib/gcc/x86_64-alpine-linux-musl/13.2.1/../../../../x86_64-alpine-linux-musl/bin/ld: cannot find -ltinfo: No such file or directory
28.39 collect2: error: ld returned 1 exit status
28.39 make: *** [Makefile:158: readsb] Error 1
```

wiedehopf, 2024-05-21:
Oh you're using sid ... complicated.

wiedehopf, 2024-05-21:
But the build logic waits for dev to have a new commit?

wiedehopf, 2024-05-21:
Anyhow can you just kick off another bump on your project, i've already put the pkg-config thing.