wiedehopf/readsb #71: FEATURE REQUEST: Prompts

mihanson, 2024-07-03:
It would be nice if the script prompted the user for SDR serial or index number and latitude/longitude when installing. It may also be helpful to remind the user to restart collectd for graphs1090 to work again.
[image]

wiedehopf, 2024-07-04:
The script isn't part of this repo.

The readsb install script readme has clear instructions on how to set the location / edit the config.

closed 2024-07-04 by wiedehopf

mihanson, 2024-07-04:
I know the readme has that info, but I thought it would be nice to enter it while installing. I understand that you do not want to implement.