wiedehopf/readsb #72: Error SDR type 'rtlsdr' not recognized

shawaj, 2024-07-04:
I have built with `dpkg-buildpackage -b -P=rtlsdr`

But I still get the below output...

```
SDR type 'rtlsdr' not recognized; supported SDR types are:
        modesbeast
        gnshulc
        ifile
        none
      ERROR: Unknown device type:rtlsdr
      Error parsing the given command line parameters, check readsb --usage and readsb --help for valid parameters.
```

Am I doing something wrong here?

shawaj, 2024-07-04:
If I try to build with `make -j3 AIRCRAFT_HASH_BITS=14 RTLSDR=yes`

Then I get this error at build time instead:

```
#13 4.676 No package 'librtlsdr' found
#13 4.753 cc -std=c11 -W -D_GNU_SOURCE -D_DEFAULT_SOURCE -Wall -Werror -fno-common -O2 -DMODES_READSB_VERSION=\""3.14.1631 wiedehopf git: 4f7a7f1 (committed: Fri Jun 14 16:52:35 2024 0200)"\" -Wdate-time -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -fstack-protector-strong -Wformat -Werror=format-security -Wno-format-truncation -DAIRCRAFT_HASH_BITS=14 -DENABLE_RTLSDR  -g  -c sdr_rtlsdr.c -o sdr_rtlsdr.o
#13 4.775 Package librtlsdr was not found in the pkg-config search path.
#13 4.775 Perhaps you should add the directory containing `librtlsdr.pc'
#13 4.775 to the PKG_CONFIG_PATH environment variable
```

wiedehopf, 2024-07-04:
<https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb>

Or just install the required packages as the readme describes?

shawaj, 2024-07-04:
I am not sure why exactly, but somewhere between these last few commits they fixed the issue...

https://github.com/NebraLtd/wingbits/commit/273b9b66e6732e9f3ba10e48f41ee0d4eb76bce6
https://github.com/NebraLtd/wingbits/commit/265fd88bb10e5e1287029068643aabbd91a26d9c
https://github.com/NebraLtd/wingbits/commit/e252704987f206ceea670a054820b2e11137cbc9

Closing the issue now as resolved, but if you have any insight into why this fixed it, maybe it can help someone else in the future :-)

closed 2024-07-04 by shawaj

wiedehopf, 2024-07-04:
Rather than making the container use an SDR ... possibly just make it get data from this for example? https://github.com/sdr-enthusiasts/docker-adsb-ultrafeeder

The most universal thing to do is probably just to get beast data and still have readsb in the container.
But it would mean you don't need to take care of the whole USB / SDR stuff.
As opposed to relying on some existing readsb json port (i believe that's what they use?)

Not a fan of anyone that promises crypto for something but have fun :)
(also their default install script would cause havoc for example with a typical dump1090-fa install ... if they would just compile / install into a directory / config that's not named readsb or anything that would be great ... just get beast data via network ... but should probably tell them not you)