wiedehopf/readsb #73: Don't use "class":"GST" for decoding location

braunschweigsailor, 2024-07-12:
Hello,

first of all, thanks for the good work on readsb and tar1090.

I'm using readsb 3.14.1631 together with tar1090 and a gps receiver. Sometimes my position on the map in tar1090 is way off, somewhere in the Atlantic Ocean off the coast of Africa, sometimes its correct.

Running readsb with --devel=debugGPS shows two different locations being decoded alternately. One being correct, the other one being wrong. The incorrect location is decoded from a GST message:

Jul 12 13:08:24 fuchsia readsb[203667]: 11:08:24.8 gpsdebug: received from GPSD: '{"class":"GST","device":"/dev/ttyUSB0","time":"2024-07-12T11:08:24.000Z","rms":3.700,"major":1.300,"minor":0.960,"orient":34.600,"lat":1.200,"lon":1.100,"alt":3.900}'
Jul 12 13:08:24 fuchsia readsb[203667]: gpsdebug: parsed lat,lon:    1.200000,   1.100000 (alt: 4 m)

When the position is correct, it was decoded from "class":"TPV".

To my understanding GST messages do not contain a location but the standard deviation of the positon and it can not be used for determining the location.

wiedehopf, 2024-07-12:
Interestingly i had code in there that ignores non-TPV messages ... not sure why i disabled that.

Please update and report back.

braunschweigsailor, 2024-07-12:
I updated and the wrong positions are gone. Thanks.

closed 2024-07-12 by wiedehopf