wiedehopf/readsb #74: Ubuntu 24.04 needs librtlsdr dependency bump

BaloneyGeek, 2024-07-13:
Hi,

Ubuntu 24.04 calls the rtl-sdr library package `librtlsdr2` instead of `librtlsdr0`. I was able to build the package fine by editing the control file locally and it runs flawlessly, so I guess it doesn't break source compatibility.

I'd submit a patch myself, but unfortunately I'm not sure how to support both package names with just one control file. Sorry about that :-(

Thanks,
Boud

wiedehopf, 2024-07-13:
Thought that was reverted to librtlsdr0 ... 
It is in debian: https://packages.debian.org/sid/libs/librtlsdr0

So i don't know what the F ubuntu is doing.
I guess they went the same way for their dev branch but didn't change it back in 24.04?
https://packages.ubuntu.com/search?keywords=librtlsdr

I'll leave this open ... but really it's not of concern to me.
Solution: don't use Ubuntu.

wiedehopf, 2024-07-13:
Or just use the containers on ubuntu ... https://sdr-enthusiasts.gitbook.io/ads-b

This version bump was a mistake and rolled back really.
Especially because the API didn't change.

jprochazka, 2024-08-21:
Came here because I noticed the same issue.

Ubuntu previously used librtlsdr0 because the package was being built against rtl-sdr version 0.6.0. They renamed it in Noble to librtlsdr2 because the package in Noble is now based on rtl-sdr version 2.0.1 code. I know what Ubuntu is doing not sure I understand why Debian would continue to label it as version 0 even though it is now using version 2 code.

wiedehopf, 2024-08-21:
So ... this fixes the build for Ubuntu24 ... i gather people will manage to do this if they find this issue.
```
sed -i -e 's/librtlsdr0/librtlsdr2/g' debian/control
```

wiedehopf, 2024-08-21:
Just as a note, the library version is usually only changed if the API changes.
Because if the API doesn't change, there is no need to change that version number exactly because it breaks dependencies.

Osmocom realized their mistake and reverted the change.
The issue is that Ubuntu did a release with this change which is very unfortunate.

jprochazka, 2024-08-21:
> Osmocom realized their mistake and reverted the change.
> The issue is that Ubuntu did a release with this change which is very unfortunate.

I see so they did set the soversion back to 0 instead of 2 in 2.0.2. It was 2 in v2.0.1 which was the version the Noble package was built against.

shawaj, 2024-08-21:
Ran into this issue running against 24.04 on GitHub runner. But have reverted to 22.04 and all good.

commit 4065eb48 (2024-08-22): fix: Ubuntu 24.04 package name change

Adds librtlsdr2 as an alternative package version if librtlsdr0 is not available. 

As per https://www.debian.org/doc/debian-policy/ch-relationships.html

Closes: #74
PR #77 (2024-08-22): fix: Ubuntu 24.04 package name change
Adds librtlsdr2 as an alternative package version if librtlsdr0 is not available. 

As per https://www.debian.org/doc/debian-policy/ch-relationships.html

Closes: #74
closed 2024-08-22 by wiedehopf