wiedehopf/readsb #75: distinct different feeders

madjuju, 2024-07-13:
Hello Merci for your great job

Is there a way to send from readsb via beast_out  a feeder ID
l explain

L have 3 instance in 3 différents localization point. They send via readsb  beast_out data to another one who display all the planes from my 3 instances.

is it possible to identify witch feeder is (source) displayed on tar1090 ? 

Sorry for my english
Julien

wiedehopf, 2024-07-13:
beast_reduce_plus_out can send a uuid ... you can compile readsb to display that uuid.

But i have no intention of supporting or explaining this more than this comment.
It's meant for aggregator purposes, not for the usecase you describe.
And it's only to debug things, not to do what you're asking.

TRACKS_UUID=yes is the compile options i think.
The aggregation instance will also have to run with --net-receiver-id (probably).

As readsb --help is hard to find:
```
--net-connector=<ip,port,beast_reduce_plus_out,uuid=k23j4k2hj34lkh23k4h23> 
```

wiedehopf, 2024-07-13:
The real trick is to just run 3 instances and look at those to gauge the coverage of each receiver.
At least that's my recommendation. (and why i don't have interest in improving the interface to display separate receivers in the same interface)

closed 2025-08-11 by wiedehopf