wiedehopf/readsb #81: [R82XX] PLL not locked! when using RTL-SDR blog V4 dongle

vatprcyq, 2024-11-29:
When inserted RTL-SDR Blog V4 dongle ,readsb show nothing received : 0.0/sec
systemctl status readsb shows:
11 29 23:52:36 DietPiM readsb[2643944]: rtlsdr: using device #0: Generic RTL2832U OEM (RTLSDRBlog, Blog V4, SN 00000001)
11 29 23:52:36 DietPiM readsb[2643944]: Detached kernel driver
11 29 23:52:37 DietPiM readsb[2643944]: Found Rafael Micro R828D tuner
11 29 23:52:37 DietPiM readsb[2643944]: rtlsdr: tuner gain set to 59.0 dB (tuner AGC)
11 29 23:52:37 DietPiM readsb[2643944]: [R82XX] PLL not locked!
11 29 23:52:37 DietPiM readsb[2643944]: [R82XX] PLL not locked!
11 29 23:52:37 DietPiM readsb[2643944]: Allocating 16 zero-copy buffers

vatprcyq, 2024-11-29:
readsb version: 3.14.1645 wiedehopf git: c3214b3 (committed: Tue Nov 19 16:39:37 2024 0100)

wiedehopf, 2024-11-29:
See notes: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb

wiedehopf, 2024-12-11:
So if the remark wasn't clear, this is likely an issue of not installing the appropriate drivers for v4.

Or bad power for the SDR or an issue with the SDR ... PLL not locked is a spurious message from librtlsdr.

closed 2024-12-11 by wiedehopf