wiedehopf/readsb #82: Invoking with --device-type and --net-only settings doesn't throw an error

shawaj, 2024-12-10:
Bit of a weird one, and probably a bit of an edge case, but if you invoke with `--device 0 --device-type rtlsdr` and also with `--net-only` set, the radio continues to work.

For example:
```
readsb --device 0 --device-type rtlsdr --gain auto --ppm 0 --max-range 450 --write-json-every 1 --write-json /run/readsb --json-location-accuracy 2 --range-outline-hours 24 --net --net-only --net-connector localhost,30006,json_out --quiet
```

I would expect in this scenario it should at least throw an error or some message in the logs / output that tells you it is preferring one setting over another. Or perhaps it shouldn't start at all due to the conflicting settings.

`readsb version: 3.14.1645`

In my case, it didn't cause an issue. I only spotted the error by accident. But reporting as I thought it might cause some unintended behaviour in the future.

wiedehopf, 2024-12-10:
--net-only is pretty much a useless option and only exists for compat.

closed 2024-12-10 by wiedehopf

shawaj, 2024-12-10:
Want me to update that in the --help or anything? 

As it currently says:
```
  --net-only                                                     Enable just networking, no RTL device or file used
```

Which I guess, based on what you are saying, is incorrect?

wiedehopf, 2024-12-10:
Well if you use contradictory cmd line options you get undefined behaviour.
Help update but i don't think it's particularly important.

NoseyNick, 2024-12-11:
> --net-only is pretty much a useless option and only exists for compat.

Wait, really? What's the correct way to run as a server, without any local SDR?

wiedehopf, 2024-12-11:
--net will turn on networking just the same as --net-only will ...

either networking or a local SDR are required for it to start up.