wiedehopf/readsb #83: Failure to build gcc 14.2.1

parkerlreed, 2025-01-05:
Arch Linux

```
rm -f readsb.pb-c.c readsb.pb-c.h
protoc-c --c_out=. readsb.proto
protoc-c --c_out=. readsb.proto
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c readsb.pb-c.c -o readsb.pb-c.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c geomag.c -o geomag.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c readsb.c -o readsb.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c anet.c -o anet.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c interactive.c -o interactive.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c mode_ac.c -o mode_ac.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c mode_s.c -o mode_s.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c comm_b.c -o comm_b.o
cc -DMODES_READSB_VERSION=\"'v4.0.4'\" -DMODES_READSB_VARIANT=\"Mictronics\" -D_GNU_SOURCE -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_PLUTOSDR -march=x86-64 -mtune=generic -O2 -pipe -fno-plt -fexceptions         -Wp,-D_FORTIFY_SOURCE=3 -Wformat -Werror=format-security         -fstack-clash-protection -fcf-protection         -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -std=c11 -O2 -g -W -D_DEFAULT_SOURCE -Wall -Werror -fno-common -Wmissing-declarations -I/usr/include/ -I/usr/include/  -c net_io.c -o net_io.o
net_io.c: In function ‘serviceInit’:
net_io.c:116:35: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
  116 |     if (!(service = calloc(sizeof (*service), 1))) {
      |                                   ^
net_io.c:116:35: note: earlier argument should specify number of elements, later size of each element
cc1: all warnings being treated as errors
make: *** [Makefile:76: net_io.o] Error 1
```

wiedehopf, 2025-01-06:
Try compiling this repository and not a different one maybe? :)

> -DMODES_READSB_VARIANT=\"Mictronics\"

This is what compiling this repo looks like:

> -DMODES_READSB_VERSION=\""3.14.1645 wiedehopf git: 48ef3c0 (committed: Sat Jan 4 19:09:58 2025 0000)"\"

Or find the correct repo to report this issue.

Looks like readsb-protobuf by mictronics ... i could link it but can't make it too easy now, can i?

closed 2025-01-06 by wiedehopf

parkerlreed, 2025-01-06:
Oh sorry about that, completely missed it, thanks!

wiedehopf, 2025-01-06:
Or if you like containers, take a look at these: https://sdr-enthusiasts.gitbook.io/ads-b/
ultrafeeder uses this readsb but there is also a container using readsb-protobuf by Mictronics.