wiedehopf/readsb #84: Issues again with librtlsdr2 on latest Ubuntu

peterould, 2025-01-13:
Upon upgrade to Ubuntu 24.10 Oracular, install of librtlsdr2 is again failing to find a suitable repository.

I believe this is a repeat of https://github.com/wiedehopf/readsb/issues/74

wiedehopf, 2025-01-13:
You're installing using which method?

```
cat /etc/os-release
```
also show this please for the scripting.

In case you're using this: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb
then please try again.

if it works now, please close ty :)

peterould, 2025-01-13:
> You're installing using which method?
> 
> ```
> cat /etc/os-release
> ```
> 
> also show this please for the scripting.
> 
> In case you're using this: https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb then please try again.
> 
> if it works now, please close ty :)

Good morning!

PRETTY_NAME="Ubuntu 24.10"
NAME="Ubuntu"
VERSION_ID="24.10"
VERSION="24.10 (Oracular Oriole)"
VERSION_CODENAME=oracular
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=oracular
LOGO=ubuntu-logo

I'm installing (updating) with the standard call

'''
sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh)"
'''

closed 2025-01-13 by peterould

wiedehopf, 2025-01-13:
Limited the special casing to use librtlsdr2 to 24.04
All other systems use librtlsdr0.

peterould, 2025-01-13:
> Limited the special casing to use librtlsdr2 to 24.04 All other systems use librtlsdr0.
> 
> (the previous implementation special cased Ubuntu 24 in general, now just specifically 24.04)

Perfect - glad to have been able to contribute in a small way to such an excellent piece of software!

</sycophancy>

peterould, 2025-03-13:
> Limited the special casing to use librtlsdr2 to 24.04 All other systems use librtlsdr0.
> 
> (the previous implementation special cased Ubuntu 24 in general, now just specifically 24.04)

Sorry to resurrect this, but getting the same issue again on Mint 22.1 Xia (testing some alternate distros) - makes sense as it's largely an Ubuntu fork

wiedehopf, 2025-03-13:
I officially don't care any more.
Not gonna fix.

peterould, 2025-03-13:
> I officially don't care any more. Not gonna fix.

Seriously though....

wiedehopf, 2025-03-13:
It's a niche issue. Being pretty serious.
You could just work around it by modifying the script.

Added a fix that will hopefully put it to rest by just trying both packages on anything that has ubuntu in /etc/os-release.

peterould, 2025-03-13:
> It's a niche issue. Being pretty serious. You could just work around it by modifying the script.
> 
> Added a fix that will hopefully put it to rest by just trying both packages on anything that has ubuntu in /etc/os-release.

I can confirm that worked on my version of Mint