wiedehopf/readsb #85: Adding --write-globe-history causes reduction in [.last1min.tracks.all] count (stats.json)

JRG1956, 2025-01-21:
Over the last month I have been seeing a significant reduction in the number of new tracks created displayed in graphs1090. (ADS-B Tracks Seen [8 minute exp. moving avg.])

The graphs1090 chart is based on data obtained from the stats.json file produced by readsb - specifically:
- last1min
   - tracks
     - all
     - single_message

By progressively removing more recently added parameters from /etc/default/readsb I identified --write-globe-history as the parameter that seems to cause the observed behaviour.

Setup used to generate results below:

[image]

All of the machines are running raspios bookworm lite 64 bit.
pi51 and pi45 are running readsb version: 3.14.1646 wiedehopf git: c373053 (committed: Fri Jan 10 17:53:36 2025 0000)
pi42 is running  v2.2.5(stable) of the ADS-B Feeder Image (I don't know how to pull the readsb version from a Docker container.)

pi42 and pi45 instances of readsb obtain their data from the instance running on pi51.

I wrote a script to run on a fourth raspberry pi that sampled the stats.json file on each of the three machines every minute and write [.last1min.tracks.all/single-message] to files for later analysis. The script obtained 128 consecutive last1min samples.

The analysis compared sums of the counts of "all tracks" and "single-message" tracks across the three machines. Pi42 does not have --write-globe-history set in any of the runs presented in the table below and acts a the "control" (Also, it isn't a parameter that is accessible through the adsb.im web interface.)

[image]

The cells shaded in orange indicate the --write-globe-history was included in the readsb configuration.

With the exception of the first run on Sunday 19th Jan, there is a consistent pattern of obtaining near matching results for machines including, or not including, this parameter.

pi51's /etc/default/readsb contents (with write-globe-history included, and exact location obscured):
```
RECEIVER_OPTIONS="--net-only --write-json-every 1"
DECODER_OPTIONS="--lat -38.29xxx --lon 144.39xxx --max-range 500 --write-globe-history /var/globe_history --write-state /var/globe_history --write-state-only-on-exit"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1280 --net-ro-interval 0.05 --net-ri-port 0 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-por>
JSON_OPTIONS="--json-location-accuracy 2 --heatmap-dir /var/globe_history --heatmap 30 --db-file /usr/local/share/tar1090/aircraft.csv.gz"
```
pi45 has the same parameters except:
```
RECEIVER_OPTIONS="--net-only --net-connector 192.168.124.108,30005,beast_in --write-json-every 1"
```
This doesn't seem to be a major issue, as it is only evident in the graphs1090 chart. There doesn't seem to be any impact on the number of positions reported.

I've documented it in case it is something you want to investigate.

wiedehopf, 2025-01-21:
This is expected behaviour.

Didn't you write about lower count of tracks on the FA forum and i asked if the configuration was different?
This was exactly why i asked.

What is a track: "Haven't seen that aircraft before."
Enabling globe history means readsb will remember aircraft much much longer.

JRG1956, 2025-01-21:
Thanks for the explanation - yes I did raise this in the FA forum and you told me to "look at the json" which is what I have done.

As far as I was concerned, the configuration wasn't different. I add comments to the config files when I make changes, including the dates. According to those notes I added the globe history parameter on November 3rd and I didn't see any change on graphs1090 until around December 23rd. Maybe I didn't restart readsb or I copied and pasted an earlier comment and forgot to change the date - I'm getting older.

Anyway, can I suggest that you edit https://github.com/wiedehopf/readsb/blob/dev/README-json.md to clarify this at some stage?

- tracks: statistics on aircraft tracks. Each track represents a unique aircraft and persists for up to 5 minutes after the last message from the aircraft is heard. If messages from the same aircraft are subsequently heard after the 5 minute period, this will be counted as a new track.
  - all: total tracks created
  - single_message: tracks consisting of only a single message. These are usually due to message decoding errors that produce a bad aircraft address.

With globe history on, clearly the 5 minutes reference doesn't hold. (When I was trying to understand the code, I found a cryptic reference to "30 or 300 seconds" in relation to new tracks - not sure what that meant.)

Also while I'm annoying you, I would suggest that you change the Y axis title on the Tracks chart to "New Tracks/Hour" as it is actually showing tracks created not total tracks in existence?

wiedehopf, 2025-01-21:
It's a terrible metric.

This whole thing makes me want to remove it like i've wanted to remove it multiple times.

JRG1956, 2025-01-21:
I was thinking the same thing, but I wasn't brave enough to suggest it.

Maybe just keep tracks with a single message - which is useful?

Could also possibly do away with the exponential smoothing and just present single messages as they occur?

Both options would probably cause outrage.

wiedehopf, 2025-01-21:
no i'm changing it to be consistent to 5 min
that's a somewhat easy change.

removing the stat would have all kinds of people complaining so that's more annoyance.

wiedehopf, 2025-01-21:
Could you check the current version?
Install script will do the trick, pushed to stale too.

JRG1956, 2025-01-21:
Thanks that seems to have fixed the issue - I updated readsb on pi51 (without globe history) and pi45 (with) and ran my script for one hour (updated around 7:00am):

pi51:  55 new tracks, 0 single message
[image]

pi45:  55 new tracks, 0 singles message
[image]

pi42:  48 new tracks, 0 singles message - is running an older version of readsb without globe history (adsb.im docker), so there may be a shift in counts between versions. I will run a couple of longer samples to check, but I don't think it matters in any case.

wiedehopf, 2025-01-21:
close enough.

The new version will be self consistent anyhow.
Even if the old version might use some non 5 minute timeframe.

As for the exponential moving average, that's just to avoid super hard spikes.
Especially when restarting it can be annoying.
The on restart spike could possibly fixed missing with collectd collection parameters.

Looking at it i don't see a way how to do it without changing the collectd DB format from DERIVE to GAUGE.
Using GAUGE has other issues.

closed 2025-01-21 by wiedehopf