wiedehopf/readsb #86: creating --ifile format?

nebukadnezar, 2025-02-08:
I seem to have a confusion: is --ifile meant to be used with raw IQ data files only? Or is there a way to dump the beast data to a file, and then read that at a later stage?

wiedehopf, 2025-02-08:
There is options to dump beast but that must be read via zstdcat | socat into beast_in wiht a special dev flag i'm not sure about right now.

--ifile is meant for what the help says, raw IQ files.

nebukadnezar, 2025-02-09:
thanks. Saving the full beast output would be a very useful option, saving as much detail as possible over a limited time period, so if anything interesting happens the data can be read back for each individual message. I'll take a stab at implementing this in my local source. If you're interested I'll share it with you.