wiedehopf/readsb #88: auto gain issues

VirusPilot, 2025-03-05:
As discussed I am experimenting with auto gain - initial results are not fully convincing. Some nearby transponder signals are not triggering any gain change at all, sometimes a reduced gain won't recover although no signal is around. I have compiled a more verbose output version but this line here won't compile: https://github.com/wiedehopf/readsb/blob/bfc4a0b83f86fc52262247efcc13a513fad62cf9/readsb.c#L830 as `action` is not defined. Can you clarify? Are there any other debug messages that I could use? The receiver is located on top of our airport tower.

wiedehopf, 2025-03-05:
Replace action with reason.

But this should be sufficient for debugging:
--gain=auto-verbose

Which gain setting are you using exactly?
Some SDRs might behave differently i suppose.

Reduced gain might not go back up if there is a lot of noise.
If there is a lot of noise, increased gain is unlikely to be useful.

VirusPilot, 2025-03-06:
I have been using standard gain settings for the most part but e.g. reduced `loudThreshold `. Still not fully understanding how all auto-gain values correspond to dBm values so that they could be adjusted based on measured values. Btw. I am using RTLSRD v3 silver sticks.

wiedehopf, 2025-03-06:
While you could translate to dB values it would still not be intuitive.

Decreasing loudThreshold seems like the correct thing to do in your situation.

Pushed a version that has that extra debug output fixed up.

> Some nearby transponder signals are not triggering any gain change at all.
Are you saying the SDR is receiving data from them? Then you really don't need to reduce gain?

VirusPilot, 2025-03-06:
I need to double check that situation where some nearby transponder signals are not triggering any gain change at all. Btw, how can I disable `tuner AGC`?

Just to confirm what the values are doing:
- noiseLowThreshold: if measured value is lower then increase gain
- noiseHighThreshold: if measured value is higher then decrease gain
- loudThreshold: if measured value is higher then decrease gain

One more question: assuming a ~30dB dynamic range of the rtlsdr, how would this translate to the 0..255 values?

VirusPilot, 2025-03-06:
just to complement, these are the values I found to work for my setup for the time being: `--gain auto-verbose,12.5,20,30,230`

wiedehopf, 2025-03-06:
tuner AGC can't be disabled as the highest gain step.
But most installs it will just switch out of that step immediately due to noise.

lowestGain is an absolute limit that autogain won't go below.

Stop asking about dB please, i already said that even if you were to calculate that it likely wouldn't make sense.

VirusPilot, 2025-03-06:
Alright, will monitor my current setting and report.

wiedehopf, 2025-07-02:
can you show the other graphs that show the worse results?
the tracks graph doesn't give any info about that.

wiedehopf, 2025-07-02:
yes it's biased towards close by, low flying planes.

as stated in the readme you can specify a minimum gain if so desired.

nikallian, 2025-07-02:
One last thing not relevant to the gain. When I switch between dump1090 and readsb the graphs relevant to the ADS-B stop working. How can I fix that ?

Thanks in advance.

RundesBalli, 2025-07-02:
> When I switch between dump1090 and readsb the graphs relevant to the ADS-B stop working. How can I fix that ?

Even if it's none of my business and I'm usually just a silent reader, I'd still like to make a rude comment:
Just don't use dump1090, because readsb is way better. ;-)

nikallian, 2025-07-03:
> Just run the graphs1090 install script after switching it will auto-detect the running decoder.
> 
> ```
> /usr/share/graphs1090/data-symlink/data
> ```
> 
> This is a symlink to either /run/readsb or /run/dump1090-fa depending on what you use. Just in case you want to manually mess with it.

Got it!

thank you very much!

wiedehopf, 2025-07-03:
Please don't ask questions about syntax if you can look at the help.
If the syntax in the help doesn't work then report it.

Also 33.8 minimum gain ... probably same as fixed gain.
I'm still curious if full autogain is actually less performance as you haven't posted any graphs that indicate it.

wiedehopf, 2025-07-03:
That's indeed not that useful because it's during a low traffic time.

The graph that spanned a week with readsb in the middle and dump1090-fa around is.... that's a useful thing to look at.
Anyhow i'm just asking for it because you said dump1090-fa was performing better.
Which it may well be.