wiedehopf/readsb #89: Stream A/C data as UDP

florian140807, 2025-04-02:
Hi
great tool, I really appreciate your job!
It would be interesting, if there could be a feature to stream the data as connection-less UDP packets (e. g. with json structured payload) Either streaming out all data, or just the one, which is selected in the map. The arguments for such command could be: all AC / selected AC, Destination IP, Destination Port.
In the documentation, I didn't see that something similar is already implement.
Sorry, if this is the wrong forum to address such questions, any advice is welcome.
Best regards, florian

wiedehopf, 2025-04-02:
Recently improved the TCP to cope better with lossy connections.

Not sure if UDP will ever be supported.
The json output is currently the closest to what you're asking for which outputs the aircraft state as json every time a new position is received for an aircraft.

florian140807, 2025-04-02:
Ok, we usually adopted UDP widely due to its simplicity. For loss detection, we have counters implemented in the payload at least to detect missing packets. And in fact, we barely miss data in the wired network with UDP.

Anyhow, could the python example here be a good starting point? https://github.com/wiedehopf/readsb/blob/dev/README-json.md ? And not printing the result, but send as UDP.

wiedehopf, 2025-04-02:
Who is we and what are you trying to do?

florian140807, 2025-04-02:
Me is I and my colleagues, we want to utilize the application to acquire the bearing information of some A/C operating around our airfield. It's mainly to get an idea, which areas (= bearing vector & distance) have better coverage than others.

wiedehopf, 2025-04-02:
Just http get the aircraft.json that's by far the easiest way.
You will even get r_dst and r_dir in the json wich is distance in nmi and bearing in degrees.

--net-json-port is TCP but it is a data stream vs a snapshot
check readsb --help, says when and what is output.

closed 2025-04-02 by wiedehopf