wiedehopf/readsb #96: Everything is running, but no data is being recorded.

Tattertots, 2025-05-12:
Updated reads to the current version from 3.14.1641. On restarting, I can see my Mode-S beast is decoding, but Readsb doesn't seem to recording anything. The aircraft.json files are empty. /tar1090 doesn't show anything either.

I don't get an error on startup

-- Boot 837dbf7bfe1441cfbb87693b1f13790a --
May 12 10:03:31 mm systemd[1]: Started readsb.service - readsb ADS-B receiver.
May 12 10:03:31 mm readsb[1366]: invoked by: /usr/bin/readsb --device 0 --device-type=modesbeast --beast-serial=/dev/ttyUSB0 --lat 1.0 --lon 1.0 --max-range 450 --write-json-every 1 --net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005 --json-location-accuracy 2 --range-outline-hours 24 --write-json /run/readsb --quiet
May 12 10:03:31 mm readsb[1366]: [2025-05-12 10:03:31.977 BST] readsb starting up.
May 12 10:03:31 mm readsb[1366]: readsb version: 3.14.1671 wiedehopf git: d280b54 (committed: Sun May 11 06:27:54 2025 0000)
May 12 10:03:31 mm readsb[1366]: Using lat: 52.226622, lon: 0.147939
May 12 10:03:31 mm readsb[1366]: 30002: Raw TCP output port
May 12 10:03:31 mm readsb[1366]: 30005: Beast TCP output port
May 12 10:03:31 mm readsb[1366]: 30003: SBS TCP output ALL port
May 12 10:03:31 mm readsb[1366]: 30001: Raw TCP input port
May 12 10:03:31 mm readsb[1366]: 30004: Beast TCP input port
May 12 10:03:31 mm readsb[1366]: 30104: Beast TCP input port
May 12 10:03:31 mm readsb[1366]: Running Mode-S Beast via USB.
May 12 10:03:31 mm readsb[1366]: 09:03:31.998 serial: creating socket client ...
May 12 10:03:31 mm readsb[1366]: 09:03:31.998 serial: creating socket client ... done

The readsb config doesn't seem to have changed... (lat/long removed)

# readsb configuration
# This is sourced by /etc/systemd/system/default.target.wants/readsb.service as
# daemon startup configuration.

RECEIVER_OPTIONS="--device 0 --device-type=modesbeast --beast-serial=/dev/ttyUSB0"
DECODER_OPTIONS="--lat 1.0 --lon 1.0 --max-range 450 --write-json-every 1"
NET_OPTIONS="--net --net-heartbeat 60 --net-ro-size 1250 --net-ro-interval 0.05 --net-ri-port 30001 --net-ro-port 30002 --net-sbs-port 30003 --net-bi-port 30004,30104 --net-bo-port 30005"
JSON_OPTIONS="--json-location-accuracy 2 --range-outline-hours 24"

Any pointers as to what to look for to diagnose?

wiedehopf, 2025-05-12:
Considering what i've seen it might just be some race condition but i don't know what.

Add
```
--devel=debugSerial
```
It doesn't matter where but just go for the receiver options.

```
--beast-baudrate=1000000
```
You could also try this. Possibly it's more stable.

But seeing it not receive anything with the debugSerial might be useful.
Possibly it's not useful though.

Tattertots, 2025-05-12:
Adding in the debugSerial has generated a large number of these... if that helps?

`May 12 11:22:05 mm readsb[21578]: 10:22:05.030 serial read return value: 1
May 12 11:22:05 mm readsb[21578]: 10:22:05.046 serial read return value: 1
May 12 11:22:05 mm readsb[21578]: 10:22:05.060 serial read return value: 3
May 12 11:22:05 mm readsb[21578]: 10:22:05.076 serial read return value: 2
May 12 11:22:05 mm readsb[21578]: 10:22:05.092 serial read return value: 4
May 12 11:22:05 mm readsb[21578]: 10:22:05.094 serial read return value: 2
May 12 11:22:05 mm readsb[21578]: 10:22:05.110 serial read return value: 3
May 12 11:22:05 mm readsb[21578]: 10:22:05.111 serial read return value: 1
May 12 11:22:05 mm578]: 10:22:05.127 serial read return value: 3
May 12 11:22:05 mm readsb[21578]: 10:22:05.143 serial read return value: 5`

wiedehopf, 2025-05-12:
almost there with the formatting
\`\`\`
code / log
\`\`\`

Ok so it's just not outputting any useful data it would seem.
Have you tried the lower baudrate?

Tattertots, 2025-05-12:
Yes, setting the lower baud rate didn't make any difference. Even tried a full power cycle, swapping out the USB cable just in case too.
It's been working for months without missing a beat.

wiedehopf, 2025-05-12:
https://github.com/wiedehopf/adsb-scripts/wiki/Automatic-installation-for-readsb#install-a-specific-version

added a readme section on how to install older versions using the script for auto install.

wiedehopf, 2025-05-12:
can you try tag=sid ?

closed 2025-05-12 by wiedehopf

Tattertots, 2025-05-12:
Confirmed, all OK without the need for specific version.

wiedehopf, 2025-05-12:
Thank you for the report and the testing.

That was very sneaky to find :)

wiedehopf, 2025-05-14:
Can you try a couple reboots + check if you get data after each reboot?
The user in the other thread is having some issues and i'd like to get some more data. (which is hard considering how few people have this device)

wiedehopf, 2025-05-14:
@Tattertots ^ forgot to mark you :)

Tattertots, 2025-05-14:
Rebooted the machine twice, came back up as expected both times.
The machine is an old Intel MacMini running Ubuntu 24.04

wiedehopf, 2025-05-14:
Yeah nevermind, it was a container version issue.
The container he was running didn't have the fix :)

Thank you for checking!

Tattertots, 2025-05-14:
No problem. Just shout if you need anything tested against a Mode-S Beast.