wiedehopf/readsb #97: Readsb "light" instance to forward to server for processing

nurunet, 2025-05-18:
Hi, so far, I had my RTL-SDR connected directly to my homelab running docker-adsb-ultrafeeder. Worked very well.

In preparation for a move where this will no longer work, I have installed readsb on a Raspberry Pi Zero 2 W to use it as an external source for ultrafeeder.

To save resources on the Pi, I am considering disabling anything not needed locally, e.g. the web interface for tar1090 (and maybe the web server?). What is the best way to do this? Simply something like "systemctl disable tar1090"?

wiedehopf, 2025-05-18:
It's not that much load really.

But yeah you can do that.

wiedehopf, 2025-05-18:
The webserver really doesn't need that much.
And even with tar1090 service disabled the webinterface keeps working (just no history).

nurunet, 2025-05-18:
Thanks a lot. I will keep an eye on the load.

closed 2025-05-18 by nurunet