wiedehopf/readsb #104: Issue re FR24 sign-up

mahonetwelve, 2025-08-10:
I had to re-do my entire setup.

So I did the following:

`sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/readsb-install.sh)"`
then
`sudo bash -c "$(curl -L -o - https://github.com/wiedehopf/graphs1090/raw/master/install.sh)"`
then
`sudo bash -c "$(wget -O - https://github.com/wiedehopf/adsb-scripts/raw/master/fr24-nopackage.sh)"`
then
`sudo fr24feed --signup; sudo systemctl restart fr24feed`
but I cannot complete the sign-up process. After enterin my existing sharing key, I get 

> Verifying sharing key...[W] Fall back to HTTP for error: SSL connect error
> Got no response or empty response from the server :(

I get the same after the lat/lon prompt when not using my sharing key.
I tried several reboots and re-takes of the setup process.

Any ideas?

Best,
Marius

wiedehopf, 2025-08-10:
yep it is <https://github.com/wiedehopf/adsb-wiki/wiki/Raspbian-Lite:-ADS-B-receiver#10-feed-fr24-optional>

Really just use adsb.im

wiedehopf, 2025-08-11:
The script error wasn inconsequential, it was simply the FR24 servers not working most likely.

The script tries to remove fr24feed as package because the package comes with a crontab that autoupdates and is problematic.