wiedehopf/readsb #107: SoapySDR unusual block size

danielmcfarland, 2025-08-27:
Hey, when I’ve compiled with soapy sdr support and attempted to use with the soapy driver however I’m repeatedly getting this error.

`weirdness: soapysdr gave us a block with an unusual size (got 65536 samples, expected 131072 samples) `

`./readsb --device-type=soapysdr --soapy-device="driver=hydrasdr"
invoked by: ./readsb --device-type=soapysdr --soapy-device=driver=hydrasdr
[2025-08-27 21:30:25.781 UTC] readsb starting up.
readsb version: 3.16.1 wiedehopf git: v3.16-5-g0cea2aa (committed: Wed Aug 27 15:11:01 2025 0000)
soapy: selected device:  driver=hydrasdr, label=HydraSDR RFOne [36b463dc31103bc7], serial=36b463dc31103bc7
soapy: hardware info:  serial=36b463dc31103bc7
soapy: driver key:      HydraSDR
soapy: hardware key:    HydraSDR
soapy: total gain:      45.0dB; LNA=15.0dB; MIX=15.0dB; VGA=15.0dB
soapy: frequency:       1090.0 MHz
soapy: sample rate:     2.4 MHz
soapy: bandwidth:       0.0 MHz
soapy: AGC mode:        manual
soapy: antenna:         RX
[INFO] Using format CS16. `

wiedehopf, 2025-08-27:
Use a lower buffer size, suppose that might be good to default to for soapy.
```
  --sdr-buffer-size=<KiB>                                        SDR buffer / USB transfer size in kibibytes (default: 256 which is equivalent to around 54 ms using rtl-sdr, option might be ignored in future versions)

```

ChrisDev8, 2025-09-21:
I've been experiencing the same error using soapysdr but lowering the buffer size doesn't appear to be fixing the issue. I'm using an SDRPlay RSP1B and the actual buffer size is always lower than what is expected.